lowRISC/ibex #1361: After updating the python version we are getting the following error

Saqlinahamad, 2021-05-25:
You are using pip version 8.1.1, however version 21.1.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
akhil@uhdct3-OptiPlex-3040:~/Saqlinahamad/lowRISC/ibex$ ^C
akhil@uhdct3-OptiPlex-3040:~/Saqlinahamad/lowRISC/ibex$ pip install --upgrade pip
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Defaulting to user installation because normal site-packages is not writeable
Requirement already up-to-date: pip in /home/akhil/.local/lib/python2.7/site-packages (20.3.4)
akhil@uhdct3-OptiPlex-3040:~/Saqlinahamad/lowRISC/ibex$ pip3 install --user -r python-requirements.txt
Collecting git+https://github.com/lowRISC/edalize.git@ot (from -r python-requirements.txt (line 6))
  Cloning https://github.com/lowRISC/edalize.git (to ot) to /tmp/pip-l79s_3rg-build
Collecting git+https://github.com/lowRISC/fusesoc.git@ot (from -r python-requirements.txt (line 9))
  Cloning https://github.com/lowRISC/fusesoc.git (to ot) to /tmp/pip-o29n9foo-build
Collecting pyyaml (from -r python-requirements.txt (line 11))
  Using cached https://files.pythonhosted.org/packages/7a/5b/bc0b5ab38247bba158504a410112b6c03f153c652734ece1849749e5f518/PyYAML-5.4.1-cp36-cp36m-manylinux1_x86_64.whl
Collecting mako (from -r python-requirements.txt (line 12))
  Using cached https://files.pythonhosted.org/packages/f3/54/dbc07fbb20865d3b78fdb7cf7fa713e2cba4f87f71100074ef2dc9f9d1f7/Mako-1.1.4-py2.py3-none-any.whl
Collecting hjson (from -r python-requirements.txt (line 15))
  Using cached https://files.pythonhosted.org/packages/74/da/f7c0e3407f1f600326ff762373f57971a16b771435736b70ce9c24fbe761/hjson-3.0.2-py3-none-any.whl
Collecting mistletoe>=0.7.2 (from -r python-requirements.txt (line 16))
  Using cached https://files.pythonhosted.org/packages/84/23/0c9c856ef73b58d8d2e40b2d028cc39cdb3a3f0b91775d4d4b7451e41f94/mistletoe-0.7.2-py3-none-any.whl
Collecting premailer (from -r python-requirements.txt (line 17))
  Using cached https://files.pythonhosted.org/packages/95/14/2f8a2e5010628a56faacd6604782b6c20180826d5529f583a9bc48f17b32/premailer-3.8.0.tar.gz
Collecting bitstring (from -r vendor/google_riscv-dv/requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/c3/fc/ffac2c199d2efe1ec5111f55efeb78f5f2972456df6939fea849f103f9f5/bitstring-3.1.7.tar.gz
Collecting Sphinx (from -r vendor/google_riscv-dv/requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/6e/8b/6662db1f132fc32e9ad10e2441a17b7d67219d7394b570434bee8b420301/Sphinx-4.0.2-py3-none-any.whl
Collecting Pallets-Sphinx-Themes (from -r vendor/google_riscv-dv/requirements.txt (line 4))
  Using cached https://files.pythonhosted.org/packages/41/cd/7f8eda22beea61237500e488cc9888f9764b3bd2b24aac0f49f2e3256c55/Pallets_Sphinx_Themes-2.0.1-py3-none-any.whl
Collecting sphinxcontrib-log-cabinet (from -r vendor/google_riscv-dv/requirements.txt (line 5))
  Using cached https://files.pythonhosted.org/packages/27/e7/dbfc155c1b4c429a9a8149032a56bfb7bab4efabc656abb24ab4619c715d/sphinxcontrib_log_cabinet-1.0.1-py2.py3-none-any.whl
Collecting sphinx-issues (from -r vendor/google_riscv-dv/requirements.txt (line 6))
  Using cached https://files.pythonhosted.org/packages/a2/cc/622a71c401b1abda6a1860ac4684e984846d1e597de24a72127ca99476d2/sphinx_issues-1.2.0-py2.py3-none-any.whl
Collecting sphinx_rtd_theme (from -r vendor/google_riscv-dv/requirements.txt (line 7))
  Using cached https://files.pythonhosted.org/packages/ac/24/2475e8f83519b54b2148d4a56eb1111f9cec630d088c3ffc214492c12107/sphinx_rtd_theme-0.5.2-py2.py3-none-any.whl
Collecting rst2pdf (from -r vendor/google_riscv-dv/requirements.txt (line 8))
  Using cached https://files.pythonhosted.org/packages/ec/b1/5e1d8d769fd2e713eda3bb75d7a46a8ef660119909f180d282df64b79ce2/rst2pdf-0.98-py3-none-any.whl
Collecting flake8 (from -r vendor/google_riscv-dv/requirements.txt (line 9))
  Using cached https://files.pythonhosted.org/packages/fc/80/35a0716e5d5101e643404dabd20f07f5528a21f3ef4032d31a49c913237b/flake8-3.9.2-py2.py3-none-any.whl
Collecting pyvsc (from -r vendor/google_riscv-dv/requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/20/48/f93d36af758ac0bdecc95474ff26c5e4428d344ecc80da7352808decbba0/pyvsc-0.3.7.20210523.1-py2.py3-none-any.whl
Collecting tabulate (from -r vendor/google_riscv-dv/requirements.txt (line 11))
  Using cached https://files.pythonhosted.org/packages/ca/80/7c0cad11bd99985cfe7c09427ee0b4f9bd6b048bd13d4ffb32c6db237dfb/tabulate-0.8.9-py3-none-any.whl
Collecting pandas (from -r vendor/google_riscv-dv/requirements.txt (line 12))
  Using cached https://files.pythonhosted.org/packages/e8/81/f7be049fe887865200a0450b137f2c574647b9154503865502cfd720ab5d/pandas-1.2.4.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-sdp9bues/pandas/setup.py", line 20, in <module>
        import numpy
    ModuleNotFoundError: No module named 'numpy'

 

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-sdp9bues/pandas/
You are using pip version 8.1.1, however version 21.1.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

imphil, 2021-05-25:
It looks like you're using Python 2.7? Only Python 3.6+ is supported: please update.

Saqlinahamad, 2021-05-25:
Okay we will update and check on that

Saqlinahamad, 2021-05-25:
For python3 -V we are getting Python 3.6.11

imphil, 2021-05-25:
But you're not using python3 in the output you've shown above. You probably need to call pip3 instead of pip.

Saqlinahamad, 2021-05-25:
I have used "pip3 install --user -r python-requirements.txt"

Saqlinahamad, 2021-05-25:
akhil@uhdct3-OptiPlex-3040:~/Saqlinahamad/lowRISC/ibex$ pip3 install --user -r python-requirements.txt
Collecting git+https://github.com/lowRISC/edalize.git@ot (from -r python-requirements.txt (line 6))
  Cloning https://github.com/lowRISC/edalize.git (to revision ot) to /tmp/pip-req-build-vylel6uo
  Running command git clone -q https://github.com/lowRISC/edalize.git /tmp/pip-req-build-vylel6uo
  Running command git checkout -b ot --track origin/ot
  Switched to a new branch 'ot'
  Branch ot set up to track remote branch ot from origin.
Collecting git+https://github.com/lowRISC/fusesoc.git@ot (from -r python-requirements.txt (line 9))
  Cloning https://github.com/lowRISC/fusesoc.git (to revision ot) to /tmp/pip-req-build-1zuzmveq
  Running command git clone -q https://github.com/lowRISC/fusesoc.git /tmp/pip-req-build-1zuzmveq
  Running command git checkout -b ot --track origin/ot
  Switched to a new branch 'ot'
  Branch ot set up to track remote branch ot from origin.
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-1zuzmveq/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-1zuzmveq/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-rdis02e7
         cwd: /tmp/pip-req-build-1zuzmveq/
    Complete output (23 lines):
    Couldn't find index page for 'setuptools_scm' (maybe misspelled?)
    No local packages or download links found for setuptools_scm
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-1zuzmveq/setup.py", line 52, in <module>
        python_requires=">=3.5, <4",
      File "/usr/lib/python3.7/distutils/core.py", line 108, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 269, in __init__
        self.fetch_build_eggs(attrs['setup_requires'])
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 313, in fetch_build_eggs
        replace_conflicting=True,
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 826, in resolve
        dist = best[req.key] = env.best_match(req, ws, installer)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1092, in best_match
        return self.obtain(req, installer)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1104, in obtain
        return installer(requirement)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 380, in fetch_build_egg
        return cmd.easy_install(req)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 657, in easy_install
        raise DistutilsError(msg)
    distutils.errors.DistutilsError: Could not find suitable distribution for Requirement.parse('setuptools_scm')
    ----------------------------------------
WARNING: Discarding git+https://github.com/lowRISC/fusesoc.git@ot. Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

imphil, 2021-05-25:
Also there is setuptools_scm vs setuptools-scm (they seem to be confused themselves). I didn't dig deeper into that yet.

closed 2021-06-14 by Saqlinahamad