lowRISC/ibex #1365: I git clone the riscv-gnu- toolchain and configured it and then i got the error as follows

AkhilBR98, 2021-05-27:
akhil@uhdct3-OptiPlex-3040:~$ cd risc/lowrisc/ibex
akhil@uhdct3-OptiPlex-3040:~/risc/lowrisc/ibex$ make build-arty-35 program-arty cd examples/sw/led && make
make[1]: Entering directory '/home/akhil/risc/lowrisc/ibex/examples/sw/led'
riscv32-unknown-elf-gcc -march=rv32imc -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -Wall -g -Os -MMD -c  -o led.o led.c
make[1]: riscv32-unknown-elf-gcc: Command not found
Makefile:50: recipe for target 'led.o' failed
make[1]: *** [led.o] Error 127
make[1]: Leaving directory '/home/akhil/risc/lowrisc/ibex/examples/sw/led'
Makefile:65: recipe for target 'examples/sw/led/led.vmem' failed
make: *** [examples/sw/led/led.vmem] Error 2
akhil@uhdct3-OptiPlex-3040:~/risc/lowrisc/ibex$

AkhilBR98, 2021-05-27:
i have used https://github.com/llowRISC/risc-gnu-toolchain this

rswarbrick, 2021-05-27:
This message means that `riscv32-unknown-elf-gcc` (part of the toolchain) is not on your path. You'll need to fix that by setting up the PATH environment variable. To check that you've done so, run `which riscv32-unknown-elf-gcc`. If it prints out the path to the binary, you're good. If not, you'll need to do more (non-Ibex-related) setup.

AkhilBR98, 2021-05-27:
But this toolchain doesn't have riscv32-unknown-elf-gcc in bin and  it has riscv64-unknown-elf-gcc so i could not add this to the path

AkhilBR98, 2021-05-27:
Can tell me which toolchain to use

rswarbrick, 2021-05-27:
Use a release from here: https://github.com/lowRISC/lowrisc-toolchains. Looks like we've got a missing item in our "System and Tool Requirements" page: I'll get to that in a minute.

AkhilBR98, 2021-05-31:
I configured riscv toolchain and then i ran the build command i got the below error
akhil@akhil-VirtualBox:~/tbex$ make build-arty-100 program-arty
cd examples/sw/led && make
make[1]: Entering directory '/home/akhil/ibex/examples/sw/led' riscv32-unknown-elf-gcc -march-rv32imc -mabi-ilp32 -static mcmodel-medany -fvis
ibility=hidden -nostdlib -nostartfiles -Wall -g -0s - MMD -c -o led.o led.c riscv32-unknown-elf-gcc -march=rv32imc -mabi-ilp32 -static mcmodel-medany -fvis
ibility=hidden -nostdlib -nostartfiles -Wall -g -Os - MMD -c -o crto.o crte.S riscv32-unknown-elf-gcc -march-rv32inc -mabi-ilp32 -static -mcmodel-medany -fvis
ibility=hidden -nostdlib -nostartfiles -Wall -g -Os -T link. ld led.o crto.o -o l
ed.elf
riscv32-unknown-elf-objcopy -0 binary led.elf led.bin srec_cat led.bin -binary offset 0x0000 byte-swap 4 -o led.vmem
make[1]: srec_cat: Command not found
Makefile:44: recipe for target 'led.vmem' failed make[1]: *** [led.vmem] Error 127
make[1]: Leaving directory '/home/akhil/ibex/examples/sw/led' Makefile:65: recipe for target 'examples/sw/led/led.vmem'
make: *** [examples/sw/led/led.vmem] Error 2
failed

imphil, 2021-05-31:
The error message says is all: "srec_cat: Command not found". Install srecord and try again.

And please, if you see error messages in the future, try to google them -- most of the time that's the most efficient way of solving the problem!

closed 2021-05-31 by imphil