lowRISC/ibex #1377: Ask a question related to Ibex

Kian75, 2021-06-07:
Hi... I have been working on Ibex core for a short time and I know that Ibex is 32IMC but I tested a code to multiply two float numbers just to make sure!! Actually I used this git hub page that use Ibex as their core: [https://github.com/pbing/ibex_wb](url) I programmed the FPGA board and it works...It actually multiplies 2.5 and 2 as float numbers and the output of multiplication is 5 (I can see the result on led)...Here is the code:
// Copyright lowRISC contributors.
// Licensed under the Apache License, Version 2.0, see LICENSE for details.
// SPDX-License-Identifier: Apache-2.0

#include <stdint.h>
#define CLK_FIXED_FREQ_HZ (25ULL * 1000 * 1000)

/**
 * Delay loop executing within 8 cycles on ibex
 */
static void delay_loop_ibex(unsigned long loops) {
  int out; /* only to notify compiler of modifications to |loops| */
  asm volatile(
      "1: nop             \n" // 1 cycle
      "   nop             \n" // 1 cycle
      "   nop             \n" // 1 cycle
      "   nop             \n" // 1 cycle
      "   addi %1, %1, -1 \n" // 1 cycle
      "   bnez %1, 1b     \n" // 3 cycles
      : "=&r" (out)
      : "0" (loops)
  );
}

static int usleep_ibex(unsigned long usec) {
  unsigned long usec_cycles;
  usec_cycles = CLK_FIXED_FREQ_HZ * usec / 1000 / 1000 / 8;

  delay_loop_ibex(usec_cycles);
  return 0;
}

static int usleep(unsigned long usec) {
  return usleep_ibex(usec);
}

int main(int argc, char **argv) {
  volatile uint32_t *var = (volatile uint32_t *) 0x10000000;
  float x=2.5;
  float y=2;
  *var = x*y;
}
After compiling I get this from compiler for the main function:
As you can see there are some F instructions....
00000000000101d0 <main>:
   101d0:	7179                	addi	sp,sp,-48
   101d2:	f422                	sd	s0,40(sp)
   101d4:	1800                	addi	s0,sp,48
   101d6:	87aa                	mv	a5,a0
   101d8:	fcb43823          	sd	a1,-48(s0)
   101dc:	fcf42e23          	sw	a5,-36(s0)
   101e0:	100007b7          	lui	a5,0x10000
   101e4:	fef43423          	sd	a5,-24(s0)
   101e8:	67c1                	lui	a5,0x10
   101ea:	4f07a787          	flw	fa5,1264(a5) # 104f0 <__errno+0x6>
   101ee:	fef42227          	fsw	fa5,-28(s0)
   101f2:	67c1                	lui	a5,0x10
   101f4:	4f47a787          	flw	fa5,1268(a5) # 104f4 <__errno+0xa>
   101f8:	fef42027          	fsw	fa5,-32(s0)
   101fc:	fe442707          	flw	fa4,-28(s0)
   10200:	fe042787          	flw	fa5,-32(s0)
   10204:	10f777d3          	fmul.s	fa5,fa4,fa5
   10208:	c01797d3          	fcvt.wu.s	a5,fa5,rtz
   1020c:	0007871b          	sext.w	a4,a5
   10210:	fe843783          	ld	a5,-24(s0)
   10214:	c398                	sw	a4,0(a5)
   10216:	4781                	li	a5,0
   10218:	853e                	mv	a0,a5
   1021a:	7422                	ld	s0,40(sp)
   1021c:	6145                	addi	sp,sp,48
   1021e:	8082                	ret
It is a bit awkward but how is it possible? How does Ibex works with this F instructions??

imphil, 2021-06-07:
No, Ibex doesn't support the F extension. Your compiler produced an instruction stream with F instructions in there, but Ibex won't be able to execute it, unless you (aka the compiler) provides an implementation for it. That's called "soft float", i.e. an implementation of floating point arithmetic in software.

Try it out by providing the corresponding "-march" flags to GCC, e.g. `-march=rv32imf` or `-march=rv32im`.

Kian75, 2021-06-08:
Thank you for your answer... I am really confused so sorry if I am asking again... 
I am trying to understand that soft float part that you mentioned and my mind is blowing :(( ... I used this following configurations:
About the compiler I used [https://github.com/riscv/riscv-gnu-toolchain](url) and I installed the (Newlib/Linux multilib)...
For compiling the code I used this make file [https://github.com/pbing/ibex_wb/blob/master/soc/fpga/arty-a7-100/sw/led/Makefile](url) in this make file the -march=rv32im ...
So you say that with this configuration for compiler Ibex can implement "soft float?!" with the help of compiler?

tomroberts-lowrisc, 2021-06-08:
> So you say that with this configuration for compiler Ibex can implement "soft float?!" with the help of compiler?

That's correct. The compiler knows that the CPU can't execute floating point arithmetic so it emulates that behavior by inserting routines based on integer arithmetic.

As a more simple example, imagine a basic CPU that doesn't have a multiply instruction. You could create a function to multiply two numbers using shift and addition instructions, then substitute that in anywhere you want to multiply. What the compiler does for soft-float is (sort-of) like that.

Kian75, 2021-06-08:
Now I get it... Thank you so much for your time and your easy to understand answer...
:))

closed 2021-06-18 by imphil