lowRISC/ibex #1378: I run the tcl script of example program in lattice with ECP5 board in lattice diamond 3.11 (synplify synthesis tool) and synthesis is exited by the error code 9 , I am not able to figure out where i had gone wrong?

Saqlinahamad, 2021-06-09:
[automake.log](https://github.com/lowRISC/ibex/files/6621582/automake.log)

GregAC, 2021-06-09:
We don't have out of the box support for ECP5 so it's not surprising a build fails. You'll need to determine how to put together an ECP5 build yourself.

I did have a quick look at your log file, this seems to be the problem:

```
@E: CG389 :"E:\Saqlain\RISCV1\ibex\build\lowrisc_ibex_top_artya7_0.1\src\lowrisc_ibex_fpga_xilinx_shared_0\rtl\fpga\xilinx\clkgen_xil7series.sv":19:7:19:17|Reference to undefined module IBUF
@E: CG389 :"E:\Saqlain\RISCV1\ibex\build\lowrisc_ibex_top_artya7_0.1\src\lowrisc_ibex_fpga_xilinx_shared_0\rtl\fpga\xilinx\clkgen_xil7series.sv":34:4:34:6|Reference to undefined module PLLE2_ADV
@E: CG389 :"E:\Saqlain\RISCV1\ibex\build\lowrisc_ibex_top_artya7_0.1\src\lowrisc_ibex_fpga_xilinx_shared_0\rtl\fpga\xilinx\clkgen_xil7series.sv":63:7:63:17|Reference to undefined module BUFG
@E: CG389 :"E:\Saqlain\RISCV1\ibex\build\lowrisc_ibex_top_artya7_0.1\src\lowrisc_ibex_fpga_xilinx_shared_0\rtl\fpga\xilinx\clkgen_xil7series.sv":68:7:68:17|Reference to undefined module BUFG
@N: CG364 :"E:\Saqlain\RISCV1\ibex\build\lowrisc_ibex_top_artya7_0.1\src\lowrisc_ibex_fpga_xilinx_shared_0\rtl\fpga\xilinx\clkgen_xil7series.sv":5:7:5:23|Synthesizing module clkgen_xil7series in library work.
```

The issue is the `clkgen_xil7series` module. This is a Xilinx specific module which instantiates a PLL to setup a clock at the desired frequency (50 MHz for our example system). You will need to replace it with something that works with the ECP5. It may be fine to just remove it entirely and use the top-level `IO_CLK` as the main clock.

Saqlinahamad, 2021-06-14:
module clkgen_xil7series (
    input IO_CLK,
    input IO_RST_N,
    output clk_sys,
    output rst_sys_n
);
  logic locked_pll;
  logic io_clk_buf;
  logic clk_50_buf;
  logic clk_50_unbuf;
  logic clk_fb_buf;
  
  buf io_clk_ibuf(io_clk_buf,IO_CLK);
  
  EHXPLLL #(
    .CLKI_DIV           (24),
    .CLKFB_DIV          (12),
    .CLKOP_DIV          (1),
    .CLKOP_ENABLE       ("ENABLED"),
    .CLKOP_CPHASE       (0),
    .CLKOP_FPHASE       (0),
    .FEEDBK_PATH        ("CLKOP"),
    .CLKOP_TRIM_POL     ("RISING"),
    .CLKOP_TRIM_DELAY   (0)
  ) pll (
    .CLKI                (io_clk_buf),
    .CLKFB               (clk_fb_buf),
    .PHASESEL1           (),
    .PHASESEL0           (),
    .PHASEDIR            (),
    .PHASESTEP           (),
    .PHASELOADREG        (),
    .STDBY               (),
    .PLLWAKESYNC         (),
    .RST                 (1'b0),
    .ENCLKOP             (),
    .ENCLKOS             (),
    .ENCLKOS2            (),
    .ENCLKOS3            (),
    .CLKOP               (clk_50_unbuf),
    .CLKOS               (),
    .CLKOS2              (),
    .CLKOS3              (),
    .LOCK                (locked_pll),
    .INTLOCK             (),
    .REFCLK              (),
    .CLKINTFB            ());

    
    // output buffering
  buf clk_fb_bufg (clk_fb_buf,clk_50_unbuf);

  buf clk_50_bufg (clk_50_buf,clk_50_unbuf);

  // outputs
  // clock
  assign clk_sys = clk_50_buf;

  // reset
  assign rst_sys_n = locked_pll & IO_RST_N;
endmodule

Saqlinahamad, 2021-06-14:
Same like PLLE2_ADV , i found EHXPLLL library in lattice but the design isn't considering the pll. the only change is i don't know the name of input buffers of lattice, which were there in xillinx like IBUF & BUFG, so i just used buf and I don't know where i had gone wrong?

GregAC, 2021-06-14:
If you want to use a PLL to alter the clock being used for Ibex you want to replace `clkgen_xil7series` entirely with something Lattice appropriate. I don't know off-hand how one uses PLLs within Lattice devices and where or if buffering will be required. It's not an Ibex support question so you'll need to look elsewhere if you have questions about it.

closed 2021-06-14 by GregAC