lowRISC/ibex #1385: regr.log doesn't contain details any more

imphil, 2021-06-18:
The regression log regr.log file used to contain information about every test run and detailed results. After the recent set of changes (some time after e716b5add84e8850074795ab9a54c190e117e8d5), which include the parallel runs, this log file doesn't contain anything but the pass/fail counts.

See https://dev.azure.com/lowrisc/lowrisc-private/_build/results?buildId=44572&view=results

Not sure if that was intended or not, but one way or the other we need a way to see which tests actually failed.

```
2021-06-18T13:00:46.5998016Z find dv/uvm/core_ibex/out -name 'regr.log' -exec cat {} \;
2021-06-18T13:00:46.6006340Z ========================== Starting Command Output ===========================
2021-06-18T13:00:46.6015989Z [command]/bin/bash --noprofile --norc /azp/agent/_work/_temp/b75f917c-f432-45bc-98b1-da41c5afd1a7.sh
2021-06-18T13:00:46.6199311Z 452 PASSED, 23 FAILED
2021-06-18T13:00:46.6428383Z ##[section]Finishing: Display regression log
```

imphil, 2021-06-18:
Thanks Rupert for the change, much better already!

https://dev.azure.com/lowrisc/4efdbdf8-a184-4518-9864-57d2a49c2aed/_apis/build/builds/44585/logs/18 is a new log file.

Unfortunately I'm greedy and want more!

We now get the details which tests failed and the type of failure ("mismatch between ISS and RTL" vs "simulation error"), but that's still less than before and not enough to know if the issue was there before, or new.

Previously we also got the simulation error (e.g. the UVM error), and a couple of mismatched instructions. The mismatched instructions aren't super useful, but the UVM error is rather helpful. See https://github.com/lowRISC/ibex/issues/1367 for an issue files based on this information.

The use case for this information: When we run the nightly regressions we know that some tests always fail. I want to be able to quickly go through the list and see which ones are almost certainly "well known", and which ones are new. Only judging by the test name is rather course-grained, having even a single line of UVM message is much better (e.g. indicating which assert fails).

Can we dump the UVM log as well for failing cases?

rswarbrick, 2021-07-12:
Hi @imphil! Greg rejigged this further as part of his recent patches. Do you have the information you need now, or is there something else that needs sorting out?

imphil, 2021-07-14:
yes, it's all good now. Here's a recent example:

```
2021-07-14T05:38:05.2013520Z ################################################################################
2021-07-14T05:38:05.2014524Z # Details of failing tests
2021-07-14T05:38:05.2015236Z ################################################################################
2021-07-14T05:38:05.2015822Z riscv_mmu_stress_test.20064
2021-07-14T05:38:05.2017342Z ---------------------------
2021-07-14T05:38:05.2018318Z Test binary: out/seed-20063/instr_gen/asm_test/riscv_mmu_stress_test_1.o
2021-07-14T05:38:05.2019399Z UVM log: out/seed-20063/rtl_sim/riscv_mmu_stress_test.20064/sim.log
2021-07-14T05:38:05.2020455Z RTL trace: out/seed-20063/rtl_sim/riscv_mmu_stress_test.20064/trace_core_00000000.log
2021-07-14T05:38:05.2021513Z ISS trace: out/seed-20063/instr_gen/spike_sim/riscv_mmu_stress_test.1.log
2021-07-14T05:38:05.2022574Z Comparison log: out/seed-20063/rtl_sim/riscv_mmu_stress_test.20064/compare.log
2021-07-14T05:38:05.2023004Z 
2021-07-14T05:38:05.2023534Z RTL / ISS trace comparison failed
2021-07-14T05:38:05.2024524Z ibex : out/seed-20063/rtl_sim/riscv_mmu_stress_test.20064/trace_core_00000000.csv
2021-07-14T05:38:05.2025833Z spike : out/seed-20063/instr_gen/spike_sim/riscv_mmu_stress_test.1.csv
2021-07-14T05:38:05.2026524Z Mismatch[1]:
2021-07-14T05:38:05.2027146Z ibex[26647] : pc[8002a4dc] rem	x17,x30,x28: a7:7ffeb000 
2021-07-14T05:38:05.2028229Z spike[26647] : pc[ffffffff80001580] addi    a5, a5, -124: a5:80047d70 
2021-07-14T05:38:05.2028872Z Mismatch[2]:
2021-07-14T05:38:05.2029452Z ibex[26648] : pc[8002a4e0] add	x9,x6,x1: s1:00000001 
2021-07-14T05:38:05.2030196Z spike[26648] : pc[ffffffff800015f0] csrr    s2, mepc: s2:8002a4d0 
2021-07-14T05:38:05.2030850Z Mismatch[3]:
2021-07-14T05:38:05.2031425Z ibex[26649] : pc[8002a4e4] xori	x30,x5,670: t5:0000029e 
2021-07-14T05:38:05.2032179Z spike[26649] : pc[ffffffff800015f4] csrr    s2, mcause: s2:00000007 
2021-07-14T05:38:05.2032843Z Mismatch[4]:
2021-07-14T05:38:05.2033466Z ibex[26650] : pc[8002a4f6] mulhsu	x12,x18,x26: a2:00000000 
2021-07-14T05:38:05.2034215Z spike[26650] : pc[ffffffff800015f8] c.li    t5, 3: t5:00000003 
2021-07-14T05:38:05.2034782Z Mismatch[5]:
2021-07-14T05:38:05.2035780Z ibex[26651] : pc[8002a4fe] lbu	x5,-1918(x3): t0:00000001 
2021-07-14T05:38:05.2036567Z spike[26651] : pc[ffffffff80001602] c.li    t5, 8: t5:00000008 
2021-07-14T05:38:05.2037146Z 266222 instructions left in trace spike
2021-07-14T05:38:05.2037674Z [FAILED]: 18118 matched, 281450 mismatch
2021-07-14T05:38:05.2037955Z 
2021-07-14T05:38:05.2038369Z riscv_illegal_instr_test.20074
2021-07-14T05:38:05.2039092Z ------------------------------
2021-07-14T05:38:05.2039976Z Test binary: out/seed-20063/instr_gen/asm_test/riscv_illegal_instr_test_11.o
2021-07-14T05:38:05.2041288Z UVM log: out/seed-20063/rtl_sim/riscv_illegal_instr_test.20074/sim.log
2021-07-14T05:38:05.2042356Z RTL trace: out/seed-20063/rtl_sim/riscv_illegal_instr_test.20074/trace_core_00000000.log
2021-07-14T05:38:05.2043471Z ISS trace: out/seed-20063/instr_gen/spike_sim/riscv_illegal_instr_test.11.log
2021-07-14T05:38:05.2045432Z Comparison log: out/seed-20063/rtl_sim/riscv_illegal_instr_test.20074/compare.log
2021-07-14T05:38:05.2045835Z 
2021-07-14T05:38:05.2046280Z RTL / ISS trace comparison failed
2021-07-14T05:38:05.2047290Z ibex : out/seed-20063/rtl_sim/riscv_illegal_instr_test.20074/trace_core_00000000.csv
2021-07-14T05:38:05.2048282Z spike : out/seed-20063/instr_gen/spike_sim/riscv_illegal_instr_test.11.csv
2021-07-14T05:38:05.2048823Z Mismatch[1]:
2021-07-14T05:38:05.2049370Z ibex[22570] : pc[80007718] csrrs	x19,0x391,x7: s3:00000000 
2021-07-14T05:38:05.2050358Z spike[22570] : pc[ffffffff80001780] addi    s6, s6, -124: s6:80027dc8 
2021-07-14T05:38:05.2050954Z Mismatch[2]:
2021-07-14T05:38:05.2051467Z ibex[22571] : pc[8000771c] c.slli	x16,0xf: a6:00020000 
2021-07-14T05:38:05.2052193Z spike[22571] : pc[ffffffff80001800] csrr    s2, mepc: s2:80007718 
2021-07-14T05:38:05.2052779Z Mismatch[3]:
2021-07-14T05:38:05.2053320Z ibex[22572] : pc[80007720] rem	x11,x20,x10: a1:00006000 
2021-07-14T05:38:05.2053985Z spike[22572] : pc[ffffffff80001804] csrr    s2, mcause: s2:00000002 
2021-07-14T05:38:05.2054556Z Mismatch[4]:
2021-07-14T05:38:05.2055088Z ibex[22573] : pc[80007724] and	x23,x23,x31: s7:86d00080 
2021-07-14T05:38:05.2055893Z spike[22573] : pc[ffffffff80001808] c.li    a1, 3: a1:00000003 
2021-07-14T05:38:05.2056564Z Mismatch[5]:
2021-07-14T05:38:05.2057163Z ibex[22574] : pc[80007730] lb	x29,724(x6): t4:ffffff86 
2021-07-14T05:38:05.2057887Z spike[22574] : pc[ffffffff80001812] c.li    a1, 8: a1:00000008 
2021-07-14T05:38:05.2058540Z 44 instructions left in trace spike
2021-07-14T05:38:05.2059077Z [FAILED]: 15561 matched, 20534 mismatch
2021-07-14T05:38:05.2059431Z 
2021-07-14T05:38:05.2059915Z riscv_debug_single_step_test.20065
2021-07-14T05:38:05.2060819Z ----------------------------------
2021-07-14T05:38:05.2061789Z Test binary: out/seed-20063/instr_gen/asm_test/riscv_debug_single_step_test_2.o
2021-07-14T05:38:05.2062902Z UVM log: out/seed-20063/rtl_sim/riscv_debug_single_step_test.20065/sim.log
2021-07-14T05:38:05.2064017Z RTL trace: out/seed-20063/rtl_sim/riscv_debug_single_step_test.20065/trace_core_00000000.log
2021-07-14T05:38:05.2064495Z 
[... 54 lines cut ...]
2021-07-14T05:38:05.2111340Z RTL trace: out/seed-20063/rtl_sim/riscv_pmp_out_of_bounds_test.20083/trace_core_00000000.log
[... cut ...]
2021-07-14T05:38:05.2124974Z ibex[2947] : pc[800192e2] lbu	x20,-1073(x2): s4:00000048 
2021-07-14T05:38:05.2125863Z spike[2947] : pc[ffffffff800017f8] c.li    a3, 3: a3:00000003 
2021-07-14T05:38:05.2126536Z Mismatch[5]:
2021-07-14T05:38:05.2127097Z ibex[2948] : pc[800192e6] lbu	x29,1753(x2): t4:00000000 
2021-07-14T05:38:05.2127826Z spike[2948] : pc[ffffffff80001802] c.li    a3, 8: a3:00000008 
2021-07-14T05:38:05.2128468Z 317364 instructions left in trace spike
2021-07-14T05:38:05.2129044Z [FAILED]: 2303 matched, 319415 mismatch
```

closed 2021-07-14 by imphil