lowRISC/ibex #1390: [dv] Azure CI runs missing log output

GregAC, 2021-06-29:
The `sim.log` file output during Azure CI runs is missing characters. This can result in false failures as the 'RISC-V UVM TEST PASSED' line isn't seen.

The artifacts from this run: https://dev.azure.com/lowrisc/lowrisc-private/_build/results?buildId=45477&view=results contain many examples. I've attached one to this issue, it's missing a `RISC-V UVM TEST PASSED' line (though also reports 0 error/fatal messages) but there's also examples of other missing bits e.g.

```
UVM_INFO /azp/agent/_work/1/s/dv/uvm/core_ibex/tests/core_ibex_seq_lib.sv(39) @ 29240000: uvm_test_top.env.irq_agent.sequencer@@irq_drop_seq_h [uvm_test_top.env.irq_agent.sequencer.irq_drop_seq_h] Starting sequence...
UVM_INFO /azp/agent/_work/1/s/dv/uvm/core_ibex/tests/core_ibex_seq_lib.sv(51) @ 29241000: uvm_test_top.env.irq_agent.sequencer@@irq_drop_seq_h [uvm_test_top.env.irq_agent.sequencer.irq_drop_seq_h] Exiting sequence
UVM_INFO /azp/agentUVM_INFO /azp/agent/_work/1/s/dv/uvm/core_ibex/tests/core_ibex_test_lib.sv(206) @ 30966000: uvm_test_top [uvm_test_top] irq_id: 0x1d
UVM_INFO /azp/agent/_work/1/s/dv/uvm/core_ibex/tests/core_ibex_test_lib.sv(330) @ 31850000: uvm_test_top [uvm_test_top] mcause: 0x8000001d
```

[sim.log](https://github.com/lowRISC/ibex/files/6732895/sim.log)

Could be a tricky one to debug, might be a VCS issue, might be an azure issue. Any thoughts?

rswarbrick, 2021-06-29:
Ahah! I can reproduce it with:
```
make -C dv/uvm/core_ibex ISS=spike IBEX_CONFIG=opentitan ITERATIONS=1 -j1 TEST=riscv_illegal_instr_test WAVES=0 COV=1 SEED=14476
```
Looks like this is caused by the coverage machinery spamming stdout at an inopportune time. I'll look in to whether we can redirect one or the other.

rswarbrick, 2021-06-29:
Right. The problem is that VCS's simv executables have a `-l` command line argument which specifies a log file. When this is specified, they write logs to both that path and stdout. We use that command line argument and also redirect stdout. To the same file. This doesn't always go well... :-) 1-line patch on the way!

PR #1392 (2021-06-29): [dv,core_ibex] Only write sim.log once
We're already redirecting stdout to sim.log in run_rtl.py. Specifying
'`-l`' as well meant that VCS opened sim.log in a separate FD.
Suprisingly enough, this mostly worked, but not always! Just write
once :-)

Fixes #1390.
closed 2021-06-29 by rswarbrick