lowRISC/ibex #1394: Makefile for RISC-V Formal does not work

ShashankVM, 2021-07-08:
## Observed Behavior

make command for risc-v formal failed

## Expected Behavior

Make command should work and I should get verilog source files to be used by RISC-V Formal
## Steps to reproduce the issue

```
$make -C formal/riscv-formal
make: Entering directory '/home/shashank/Downloads/ibex-master/formal/riscv-formal'
make: *** No rule to make target '../rtl/ibex_alu.sv', needed by 'build/ibex_alu.v'.  Stop.
make: Leaving directory '/home/shashank/Downloads/ibex-master/formal/riscv-formal'
```

## My Environment
Dowloaded zip file from ibex github

**Operating system:**
Kubuntu 21.04

**Version of the Ibex source code:**
https://github.com/lowRISC/ibex/commit/90ff7ca6c3367da721dba13e91dfc98f782a5745

rswarbrick, 2021-07-12:
Hi @ShashankVM! Thanks for the bug report. Yep, it's definitely broken! I've spent a little bit of time looking into this today and I think we could probably get it synthesizing with Yosys again. However, I'm not sure whether that would be much use: this has bit-rotted because no-one was using it, so it's possible that there's something fundamental that has broken in the meantime.

Do you have a use-case for this? Otherwise, it might actually be better to remove it entirely.

ShashankVM, 2021-07-12:
Hi @rswarbrick, yes, there is an important use case for this -  formal verification. Do you have a formal verification testbench as comprehensive as RISC-V Formal? If so, there is no problem in removing this one. PS: It's also okay if you use the licensed version of Symbiyosys (which is free for open-source projects) and support formal verification on the DUT in SystemVerilog, instead of converting to Verilog as it is currently done.

rswarbrick, 2021-07-12:
Ok, if you're willing to do some further debug, I'll push up some patches that get the flow working. See #1399 for the PR. Let us know how the testing goes!

ShashankVM, 2021-07-12:
@rswarbrick Thank you. I will definitely check this sometime and report back.

towoe, 2021-07-13:
@ShashankVM you can use https://github.com/towoe/riscv-formal/tree/ibex-top as a staring point if you want.
It uses the current `ibex_top` module.
The wrapper would still need some work as some checks are failing, but the flow to run the checks should work.

closed 2021-07-14 by rswarbrick