lowRISC/ibex #1405: Ask a question related to Ibex

Kian75, 2021-07-19:
Hi I am working on Ibex recently and I have written a Convolutional Neural Network (CNN)  in C to implement it on Ibex. First the network was very small and it worked pretty well on the core, but now that it is fairly a medium size network core is not working (I want to light an led corresponding to the class of input). It may be silly to ask but as we generate the .mem file for the core is there any restriction for the generated lines to be executed on the core or I should look to other things to find my issue? For my project I used [https://github.com/pbing/ibex_wb](url) and the compiler is [https://github.com/riscv/riscv-gnu-toolchain](url)

rswarbrick, 2021-07-21:
Hi there! The Ibex core doesn't really have a restriction on the size of the data and instruction memories that it's connected to (except for a 32-bit address space, giving 4GiB). But if you're using this in a real system, there will be practical limitations, governed by the size of the memory on your chip (assuming no external RAM).

However, the style of the question makes me think that you're writing software for Ibex, not dealing with the details of how to integrate the core into a larger system. Are you using the simple_system demo wrapper? That has a 1MiB RAM: maybe check that your code fits?

Kian75, 2021-07-21:
Thank you for your answer.... 
Yes I am writing a software for Ibex....Actually it fits on FPGA and takes about 88% of the BRAMs. In [https://github.com/pbing/ibex_wb/tree/master/soc/fpga/arty-a7-100/sw/led](url) they provided a file name link.ld which indicates how much memory you want to have for the core, and after compiling it says that is the memory enough for the written code or not. Anyway.... I tested smaller version of my CNN(Changed the input image size from 150*150 to 50*50) and it worked with a little delay for lighting the LED but when I changed it into 100*100 it didn't work again. How can I justify that? Should I say due to lots of data in my code the core cannot process it or what?

rswarbrick, 2021-07-21:
I don't think we can really help you further here: your question now looks an awful lot like "I've written some code and it doesn't work. Why?" :smile: I'd suggest you make sure that your code and data fit in the memory you have and then, if it still doesn't work, try to reproduce the failure in Spike.

If you're completely stuck, the first place to ask is probably the system that integrates everything, which appears to be the `ibex_wb` project. I'm closing this issue now: please feel free to re-open if you've got some data that suggests the Ibex core isn't working properly.

closed 2021-07-21 by rswarbrick