lowRISC/ibex #1413: Support of dynamic memory allocation functions on Ibex

Kian75, 2021-07-29:
Hi I am new to ibex and probably my question is a bit low level too but I have been working to write a C code on Ibex and I realized because of limited resources on my fpga board I have to allocate and deallocate memory using malloc and free functions. I wrote a simple code first but when I implemented it on the fpga it didn't work and the code stock before malloc function (I realized it by using led to debug the code on fpga) . So is there any problem in my code or malloc and free function is not supported on Ibex?
Here is the code:

// Copyright lowRISC contributors.
// Licensed under the Apache License, Version 2.0, see LICENSE for details.
// SPDX-License-Identifier: Apache-2.0

#include <stdint.h>
#include <stdlib.h>
#define CLK_FIXED_FREQ_HZ (25ULL * 1000 * 1000)

/**
 * Delay loop executing within 8 cycles on ibex
 */
static void delay_loop_ibex(unsigned long loops) {
    int out; /* only to notify compiler of modifications to |loops| */
    asm volatile(
        "1: nop             \n" // 1 cycle
        "   nop             \n" // 1 cycle
        "   nop             \n" // 1 cycle
        "   nop             \n" // 1 cycle
        "   addi %1, %1, -1 \n" // 1 cycle
        "   bnez %1, 1b     \n" // 3 cycles
        : "=&r" (out)
        : "0" (loops)
        );
}

static int usleep_ibex(unsigned long usec) {
    unsigned long usec_cycles;
    usec_cycles = CLK_FIXED_FREQ_HZ * usec / 1000 / 1000 / 8;
    delay_loop_ibex(usec_cycles);
    return 0;
}

static int usleep(unsigned long usec) {
    return usleep_ibex(usec);
}

int main(int argc, char **argv) {
    volatile uint32_t *var = (volatile uint32_t *) 0x10000000;
    int i,sum=0;
    *var = 1;
    uint32_t *ptr = malloc(5 * sizeof(uint32_t));
    *var = 2;
    for(i=0;i<5;i++){
        * (ptr+i)= i * 10;
    }
    *var = 3;
    for(i=0;i<5;i++){
        sum += * (ptr+i);
    }
    *var = sum;
    free(ptr);
    ptr=NULL;
}

So on fpga only led number 1 is lighting up.

rswarbrick, 2021-07-29:
My guess is that the libc that you're linking against has the wrong view of memory and its heap ends up using some addresses that don't exist. Have you tried running this on an instruction set simulator like Spike? You could take a look at the trace and I suspect you'll see writes to bogus addresses.

Kian75, 2021-07-29:
Thank you for your answer...
Sorry but I am very new in Ibex and RiscV environment... I can send the linker file but about the libc I actually don't know what that is :(( ... Can you explain more please? Or if you can share any reference so that I can look myself? Sorry that I am bothering you...

Here is link.ld:
OUTPUT_ARCH(riscv)

/* required to correctly link newlib */
GROUP( -lc -lgloss -lgcc -lsupc++ )

SEARCH_DIR(.)
__DYNAMIC  =  0;

MEMORY
{
    rom         : ORIGIN = 0x00000000, LENGTH = 0x36800 /* 218 kB */
    stack       : ORIGIN = 0x00036800, LENGTH = 0x4F000 /* 320 kB */
}

/* Stack information variables */
_min_stack      = 0x2000;   /* 8K - minimum stack space to reserve */
_stack_len     = LENGTH(stack);
_stack_start   = ORIGIN(stack) + LENGTH(stack);

/* We have to align each sector to word boundaries as our current s19->slm
 * conversion scripts are not able to handle non-word aligned sections. */

SECTIONS
{
    .vectors :
    {
        . = ALIGN(4);
        KEEP(*(.vectors))
    } > rom

    .text : {
        . = ALIGN(4);
        _stext = .;
        *(.text)
        *(.text.*)
        _etext  =  .;
        __CTOR_LIST__ = .;
        LONG((__CTOR_END__ - __CTOR_LIST__) / 4 - 2)
        *(.ctors)
        LONG(0)
        __CTOR_END__ = .;
        __DTOR_LIST__ = .;
        LONG((__DTOR_END__ - __DTOR_LIST__) / 4 - 2)
        *(.dtors)
        LONG(0)
        __DTOR_END__ = .;
        *(.lit)
        *(.shdata)
        . = ALIGN(4);
        _endtext = .;
    }  > rom

    .rodata : {
        . = ALIGN(4);
        *(.rodata);
        *(.rodata.*)
    } > rom

    .shbss :
    {
        . = ALIGN(4);
        *(.shbss)
    } > rom

    .data : {
        . = ALIGN(4);
        sdata  =  .;
        _sdata  =  .;
        *(.data);
        *(.data.*)
        edata  =  .;
        _edata  =  .;
    } > rom

    .bss :
    {
        . = ALIGN(4);
        _bss_start = .;
        *(.bss)
        *(.bss.*)
        *(.sbss)
        *(.sbss.*)
        *(COMMON)
        _bss_end = .;
    } > rom

    /* ensure there is enough room for stack */
    .stack (NOLOAD): {
        . = ALIGN(4);
        . = . + _min_stack ;
        . = ALIGN(4);
        stack = . ;
        _stack = . ;
    } > stack

    .stab  0 (NOLOAD) :
    {
        [ .stab ]
    }

    .stabstr  0 (NOLOAD) :
    {
        [ .stabstr ]
    }
}

rswarbrick, 2021-07-29:
I don't think I'm going to be able to debug this for you, but I did notice this in your linker script:
```
MEMORY
{
rom : ORIGIN = 0x00000000, LENGTH = 0x36800 /* 218 kB /
stack : ORIGIN = 0x00036800, LENGTH = 0x4F000 / 320 kB */
}
```
The only writable data memory is called "stack". Looks like this isn't supposed to support a heap at all.

Kian75, 2021-07-29:
Oh... ok so do you know what can I do about it? because I need to free allocated space because of shortage of brams on my fpga
Should I add sth to this linker file??

vogelpi, 2021-09-01:
I am closing this issue as the original question has been answered and a new follow-up issue has been created here: #1415
This one also contains relevant pointers on how to implement dynamic memory allocation on embedded systems.

closed 2021-09-01 by vogelpi