lowRISC/ibex #1415: How to add heap memory in link.ld file

Kian75, 2021-07-31:
Hi... I am working on Ibex recently (sorry that I am asking lots of questions from you guys) and I want to use dynamic memory allocation functions in C... So I have to add heap memory in linker file... How can I do that?
Here is the example link.ld file that you have provided in your examples:

OUTPUT_ARCH(riscv)

/* required to correctly link newlib */
GROUP( -lc -lgloss -lgcc -lsupc++ )

SEARCH_DIR(.)
__DYNAMIC  =  0;

MEMORY
{
    rom         : ORIGIN = 0x00000000, LENGTH = 0xC000 /* 48 kB */
    stack       : ORIGIN = 0x0000C000, LENGTH = 0x4000 /* 16 kB */
}

/* Stack information variables */
_min_stack      = 0x2000;   /* 8K - minimum stack space to reserve */
_stack_len     = LENGTH(stack);
_stack_start   = ORIGIN(stack) + LENGTH(stack);

/* We have to align each sector to word boundaries as our current s19->slm
 * conversion scripts are not able to handle non-word aligned sections. */

SECTIONS
{
    .vectors :
    {
        . = ALIGN(4);
        KEEP(*(.vectors))
    } > rom

    .text : {
        . = ALIGN(4);
        _stext = .;
        *(.text)
        *(.text.*)
        _etext  =  .;
        __CTOR_LIST__ = .;
        LONG((__CTOR_END__ - __CTOR_LIST__) / 4 - 2)
        *(.ctors)
        LONG(0)
        __CTOR_END__ = .;
        __DTOR_LIST__ = .;
        LONG((__DTOR_END__ - __DTOR_LIST__) / 4 - 2)
        *(.dtors)
        LONG(0)
        __DTOR_END__ = .;
        *(.lit)
        *(.shdata)
        . = ALIGN(4);
        _endtext = .;
    }  > rom

    .rodata : {
        . = ALIGN(4);
        *(.rodata);
        *(.rodata.*)
    } > rom

    .shbss :
    {
        . = ALIGN(4);
        *(.shbss)
    } > rom

    .data : {
        . = ALIGN(4);
        sdata  =  .;
        _sdata  =  .;
        *(.data);
        *(.data.*)
        edata  =  .;
        _edata  =  .;
    } > rom

    .bss :
    {
        . = ALIGN(4);
        _bss_start = .;
        *(.bss)
        *(.bss.*)
        *(.sbss)
        *(.sbss.*)
        *(COMMON)
        _bss_end = .;
    } > rom

    /* ensure there is enough room for stack */
    .stack (NOLOAD): {
        . = ALIGN(4);
        . = . + _min_stack ;
        . = ALIGN(4);
        stack = . ;
        _stack = . ;
    } > stack

    .stab  0 (NOLOAD) :
    {
        [ .stab ]
    }

    .stabstr  0 (NOLOAD) :
    {
        [ .stabstr ]
    }
}
What should I change here in order to have heap memory?

ErfanElec, 2021-08-01:
Hello everyone,

This is my problem as well. I have implemented Ibex on Nexys 4 DDR for my image processing project. I ran out of memory, hence, I need to add a heap memory in the linker file in order to manage the memory. Can you guide us on how to add heap memory to the current project?

imphil, 2021-08-02:
>  I ran out of memory, hence, I need to add a heap memory in the linker file in order to manage the memory.

Not necessarily. You can manually manage memory without malloc()/free(), and for very regular programs like image processing that's actually a reasonably common and good approach.

I'm wondering: are you doing a university project or something of that sort? In that case, it might be best to talk to your adviser there and ask them to explain to you how memory allocation on embedded systems work, i.e. how malloc() and free() are implemented, and how C libraries (such as newlib) work and can be customized.

A couple points for further debugging:
- Figure out what your code is doing when it calls malloc(). Look at the trace that ibex produces, match it against the binary, and see where in the underlying C code you're hanging.
- You're probably using newlib as C library. Have a look at the description in the newlib source code how malloc() works there: https://sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=newlib/libc/stdlib/malloc.c;h=f5ac2920888563013663454758cce102e40b69ad;hb=2b50ec0cd205c72b17d8a905f7cfd9e7bb3a6794#l118
- It is likely you'll need to implement `_sbrk()`, or provide appropriate linker symbols for the default implementation to work. A couple of pointers to potentially helpful discussions:
	- https://sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=libgloss/riscv/sys_sbrk.c;h=086509efa0bb9efd42f6555d0cd4dadcdd73354c;hb=2b50ec0cd205c72b17d8a905f7cfd9e7bb3a6794
	- https://github.com/riscv/riscv-gnu-toolchain/issues/559
	- https://github.com/riscv/riscv-gnu-toolchain/issues/571

closed 2021-12-10 by GregAC