lowRISC/ibex #1423: Implenting implementer specified reset value for ePMP

GregAC, 2021-08-23:
OpenTitan will want a PMP region configured on reset, along with a non-zero reset value of `mseccfg`: https://github.com/lowRISC/opentitan/issues/7834

How do we want to support this in Ibex? I don't think we should use top-level parameters as it'll be a rarely used feature and would require many parameters. We could use `define`. Perhaps if a single `IBEX_CUSTOM_PMP_RESET_VAL` is defined it then uses a whole set of `IBEX_CUSTOM_PMPCFG0_RESET_VAL` `IBEX_CUSTOM_PMPADDR0_RESET_VAL` etc then specify all of the reset values.

Another possibility is an `ibex_pmp_reset.svh` include. It'd have a bunch of `localparam` giving all of the reset values. We'd supply a default one with Ibex (that just resets to 0) and implementers, such as OpenTitan, would supply a different include. This may mean a patch against Ibex when we vendor into OT, or perhaps we could use a `define` to specify the name of the include file? Maybe there's something we can do with fusesoc to override the include file used in the build (@imphil any cunning ideas)?

vogelpi, 2021-09-01:
I agree that we should use parameters to propagate the reset values and I think we don't want to use defines either. The idea of an include file sounds good to me. Or we could just specify the reset values as `localparam/parameters` inside `ibex_pkg.sv` and then ask implementers to patch that when vendoring in Ibex. This is already supported by the vendor tool.

GregAC, 2021-09-01:
I'd prefer a separate file for the parameters for this to simplify vendoring. An `ibex_pmp_reset.svh` file should see minimal/no change so a patch for it should just keep working.

imphil, 2021-09-01:
I'm not really a fan of patching, it's a very unreliable extension vector for something we actually want to support.

So a bit of define logic is probably needed unless we want to put the logic into the build system, which is not trivial to do in FuseSoC, and doesn't help our users who don't use FuseSoC.

So how about something like:

```
`ifdef IBEX_CUSTOM_PMP_RESET_VALUES
`include "ibex_pmp_reset.svh"
`else
parameter Pmp0ResetValue = 'd12;
...
`endif
```

It's a single define that we need to support, and then users can simply prepare a file and put it into the include path. For OpenTitan, we can put the file into the rv_core_ibex.core (https://github.com/lowRISC/opentitan/blob/master/hw/ip/rv_core_ibex/rv_core_ibex.core).

rswarbrick, 2021-09-01:
I agree with @imphil, although I'd consider putting the default values into an `ibex_pmp_reset_default.svh` and including that in the `else` branch.

PR #1512 (2022-01-24): [rtl,doc] Add customisable PMP reset values
Fixes #1423
commit 97fa5cf2 (2022-01-24): [rtl,doc] Add customisable PMP reset values

Fixes #1423
closed 2022-01-24 by GregAC