lowRISC/ibex #1425: Report a bug in Ibex

nicolast0604, 2021-08-25:
[Synth 8-3438] module 'BUFGCE' declared at '/tools/Xilinx/Vivado/2018.3/scripts/rt/data/unisim_comp.v:619' does not have any parameter 'SIM_DEVICE' used as named parameter override ["/home/nioclast0604/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_prim_xilinx_clock_gating_0/rtl/prim_xilinx_clock_gating.sv":18]
[Synth 8-6156] failed synthesizing module 'prim_xilinx_clock_gating' ["/home/nioclast0604/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_prim_xilinx_clock_gating_0/rtl/prim_xilinx_clock_gating.sv":5]
[Vivado_Tcl 4-5] Elaboration failed - please see the console for details

## Observed Behavior

Vivado synthesis & implementation for Arty A7 FPGA board
## Expected Behavior

bitstream finished
## Steps to reproduce the issue
make build-arty-35 program-arty

## My Environment

**EDA tool and version:**
Vivado 2018.3

**Operating system:**

Ubuntu 18.04.5
**Version of the Ibex source code:**

master fork 246

vogelpi, 2021-08-25:
Hi @nicolast0604 ,

thanks for reporting this issue. It seems that Vivado 2018.3 doesn't support the `SIM_DEVICE` parameter on the `BUFGCE` instance. We should probably recommend the use of a newer Vivado version now.

As a local workaround, you can do the following change inside `ibex/vendor/lowrisc_ip/ip/prim_xilinx/rtl/prim_xilinx_clock_gating.sv`:
```sv
    BUFGCE #(
      .SIM_DEVICE("7SERIES")
    ) u_bufgce (
```
to
```sv
    BUFGCE u_bufgce (
```

vogelpi, 2021-08-31:
I am closing this issue as we now require Vivado 2020.2 which understands this parameter.

closed 2021-08-31 by vogelpi