lowRISC/ibex #1432: I wanted to connect BRAM with AXI4 spec for artya7 example instead of SRAM/with SRAM , can I be able to connect it ?

Saqlinahamad, 2021-09-07:
Had facing the understanding issue
As I am not able to figure out how the ibex core is transferring the instructions to write the data to SRAM, 
(As in example says the LEDs are blinked ,for a word transfer DATA[8*i +:4] )
Please help me in understanding how to connect BRAM with AXI4 spec for artya7 ex instead of SRAM/with SRAM.

GregAC, 2021-09-10:
Hi @Saqlinahamad you'll need to translate Ibex's memory interface into an AXI4 interface. We don't provide anything to do this however Ibex's interface is pretty simple so it shouldn't be too complex. There's details of the protocol here: https://ibex-core.readthedocs.io/en/latest/03_reference/load_store_unit.html#protocol

closed 2021-09-13 by Saqlinahamad

brankob89, 2021-10-07:
Hi @GregAC, so for APB, AHB, AXI, we should have some bridge from ibex mem interface to some other protocol? And can it be directly transfer to tilelink interface?