lowRISC/ibex #1437: Getting Zeros at memory regions !!

KinzaQamar, 2021-09-19:
I got a test failing on seed 27237 , but when i check the .S file , both the regions got zeros inside them.
The test was riscv-mmu-stress-test

rswarbrick, 2021-09-20:
Can you give the exact command you ran to see the error? Also please can you give us the version of the repository that you're using: you can get it by running `git describe --dirty --always`.

KinzaQamar, 2021-09-20:
The version of the repo that I am runing is "86084b9-dirty"

KinzaQamar, 2021-09-20:
one more thing I discover today is that when I ran 10 iterations for the same test,while all 10 were passing , I got an assembly file with zeros in regions again.
This time, on passing seed !!

rswarbrick, 2021-09-20:
The `-dirty` suffix above means you've got local modifications to your repository. Please can you try to reproduce the error with a clean checkout. Then please post the exact command line you ran.

KinzaQamar, 2021-09-20:
How can I achieve clean checkout ?

GregAC, 2021-09-20:
Note that the Ibex regression isn't entirely clean, `riscv-mmu-stress-test` has been know to have failures. These tend to be testbench issues rather than RTL problems.

We're in the process of updating our testbench to use a co-simulation approach (https://github.com/lowRISC/ibex/pull/1416). Once that work is complete we will put more effort into fixing up current regression issues.

GregAC, 2021-09-20:
> How can I achieve clean checkout ?

You can use `git stash` (https://git-scm.com/docs/git-stash) to save your local changes. If you're happy to discard local changes a `git reset --hard` will do the job (WARNING: Don't do this unless you're happy with loosing local file changes). You can also use `git clean` to remove untracked files, but that's probably not necessary. This article: https://remarkablemark.org/blog/2018/10/09/git-checkout-clean-reset/ looks to give a reasonable overview of clean and reset.

KinzaQamar, 2021-09-20:
Hey GregAC !
May you please answer that issue too (https://github.com/lowRISC/ibex/issues/1438#issuecomment-922996643)

closed 2021-10-02 by KinzaQamar