lowRISC/ibex #1440: Error when running ./syn_yosys.sh

Merok, 2021-09-22:
## Observed Behavior

Hello,
I am new to Ibex and try to experiment with it.
I have an issue when running the ./syn_yosys.sh script. 
More precisely, I have such error: 
`Executing Verilog-2005 frontend: syn_out/ibex_22_09_2021_14_37_02/generated/ibex_top_tracing.v
syn_out/ibex_22_09_2021_14_37_02/generated/ibex_top_tracing.v:99: ERROR: FATAL: Fatal error: RVFI needs to be defined globally..
Failed to synthesize RTL with Yosys`

I did define the syn_setup.sh script with the Nangate45 lib.

Do you have an idea of why I get such error ?

Thanks for your help.

## My Environment

**Operating system:**

I am working on Ubuntu 20.04.3 amd64.
**Version of the Ibex source code:**

I am using the ibex source code version ac8934459b028220ce9fd20683bb018244b59756.

rswarbrick, 2021-09-22:
@GregAC: I think you know most about our synthesis flow?

Merok, 2021-09-27:
Hello,
Do you have some updates for my problem ?

Thanks a lot!

tomroberts-lowrisc, 2021-09-27:
Can you let us know what versions of yosys and sv2v you are using? The script works locally for me using yosys v0.9 and sv2v v0.0.5-21-g6d907e0

Merok, 2021-09-27:
Here are the versions I use:

* Yosys 0.9+4081 (git sha1 UNKNOWN, gcc 8.4.1 -O2 -fexceptions -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fPIC -Os)
* sv2v v0.0.8-42-g814f965

I tried on another machine on Rocky Linux 8.4. I have the same error.
Here are my steps:
* git clone https://github.com/lowRISC/ibex.
* I installed the requirements in https://github.com/lowRISC/ibex/tree/master/syn.
* I downloaded the nangate45.lib (https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/tree/master/flow/platforms/nangate45/lib)
* I uncommented and modified the two last lines of the syn_setup.sh file such as:
```
export LR_SYNTH_CELL_LIBRARY_PATH=/test/x/Nangate45/NangateOpenCellLibrary_typical.lib
export LR_SYNTH_CELL_LIBRARY_NAME=nangate
```
* I go to the the syn directory and run syn_yosys.sh

here is the full output:

```
******  o p e n   t o   t h e   c o r e  *******

       +---------------------------------+
       |  _   _   ___   ___  _   _  ___  |
       | | | | | / _ \ / __|| | | |/ __| |
       | | |_| || (_) |\__ \| |_| |\__ \ |
       |  \__, | \___/ |___/ \__, ||___/ |
       |   __/ |              __/ |      |
       |  |___/              |___/       |
       +---------------------------------+

=================== Flow Vars ===================
Path to cell library: /XX/Nangate45/NangateOpenCellLibrary_typical.lib
top module: ibex_top (default)
Output directory for synthesis: syn_out/ibex_27_09_2021_16_54_01
Pre-mapping netlist out: ./syn_out/ibex_27_09_2021_16_54_01/generated/ibex_top.pre_map.v (default)
netlist out: ./syn_out/ibex_27_09_2021_16_54_01/generated/ibex_top_netlist.v (default)
Synth config file: ibex_top_lr_synth_conf.tcl (default)
Report output directory: ./syn_out/ibex_27_09_2021_16_54_01/reports (default)
flatten: Enabled
timing run: Enabled
Enable branch target ALU in Ibex: Disabled
Enable writeback stage in Ibex: Disabled
Bitmanip extenion setting for Ibex (see ibex_pkg::rv32b_e for permitted values. Enum names are not supported in Yosys.): 0 (default)
Multiplier extension setting for Ibex (see ibex_pkg::rv32m_e for permitted values. Enum names are not supported in Yosys.): 2 (default)
Register file implementation selection for Ibex (see ibex_pkg::regfile_e for permitted values. Enum names are not supported in Yosys.): 2 (default)
Name of cell library: nangate
Input SDC file: ibex_top.nangate.sdc (default)
Input SDC file for ABC: ibex_top_abc.nangate.sdc (default)
Output SDC file: ./syn_out/ibex_27_09_2021_16_54_01/generated/ibex_top.nangate.out.sdc (default)
STA netlist out: ./syn_out/ibex_27_09_2021_16_54_01/generated/ibex_top_netlist.sta.v (default)
STA paths reported per group: 1000 (default)
STA paths reported in overall report: 1000 (default)
clock period: 4000.0 ps
=================================================

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Wolf <claire@symbioticeda.com>          |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
[... 118 lines cut ...]
22. Executing Verilog-2005 frontend: syn_out/ibex_27_09_2021_16_54_01/generated/ibex_pmp.v
Parsing SystemVerilog input from `syn_out/ibex_27_09_2021_16_54_01/generated/ibex_pmp.v' to AST representation.
Generating RTLIL representation for module `\ibex_pmp'.
Successfully finished Verilog frontend.

23. Executing Verilog-2005 frontend: syn_out/ibex_27_09_2021_16_54_01/generated/ibex_prefetch_buffer.v
Parsing SystemVerilog input from `syn_out/ibex_27_09_2021_16_54_01/generated/ibex_prefetch_buffer.v' to AST representation.
Generating RTLIL representation for module `\ibex_prefetch_buffer'.
Successfully finished Verilog frontend.

24. Executing Verilog-2005 frontend: syn_out/ibex_27_09_2021_16_54_01/generated/ibex_register_file_latch.v
Parsing SystemVerilog input from `syn_out/ibex_27_09_2021_16_54_01/generated/ibex_register_file_latch.v' to AST representation.
Generating RTLIL representation for module `\ibex_register_file_latch'.
Warning: Replacing memory \mem with list of registers. See syn_out/ibex_27_09_2021_16_54_01/generated/ibex_register_file_latch.v:113, syn_out/ibex_27_09_2021_16_54_01/generated/ibex_register_file_latch.v:87
Successfully finished Verilog frontend.

25. Executing Verilog-2005 frontend: syn_out/ibex_27_09_2021_16_54_01/generated/ibex_top.v
Parsing SystemVerilog input from `syn_out/ibex_27_09_2021_16_54_01/generated/ibex_top.v' to AST representation.
Generating RTLIL representation for module `\ibex_top'.
Successfully finished Verilog frontend.

26. Executing Verilog-2005 frontend: syn_out/ibex_27_09_2021_16_54_01/generated/ibex_top_tracing.v
syn_out/ibex_27_09_2021_16_54_01/generated/ibex_top_tracing.v:99: ERROR: FATAL: Fatal error: RVFI needs to be defined globally..
Failed to synthesize RTL with Yosys
```

Merok, 2021-10-01:
Hi, 
Any help on this error would be greatly appreciated.

Thanks.

imphil, 2021-10-01:
Please try using the exact same versions of yosys and sv2v as @tomroberts-lowrisc mentioned above; the syn_yosys.sh scripting is highly fragile (and unsupported), so if you'd like to use other versions you'll probably need to do some digging yourself.

GregAC, 2021-10-05:
I can also run `syn_yosys.sh` without issue. I'm using an older version of sv2v and yosys (sv2v v0.0.5-18-ga170536 and Yosys 0.9+1706 (git sha1 968956ba, clang 6.0.0-1ubuntu2 -fPIC -Os). It looks like either the newer sv2v is translating things a bit differently and yosys doesn't like it or sv2v is emitting some slightly broken verilog that previously yosys didn't care about but now does.

Merok, 2021-10-06:
Hi,

Thanks for the help.
Indeed, the ./syn_yosis.sh script works now with sv2v v0.0.5-18-ga170536.

closed 2021-10-06 by Merok

zachjs, 2021-10-06:
Please try out https://github.com/lowRISC/ibex/pull/1449, which should make the flow work with both older and newer versions of sv2v.