lowRISC/ibex #1441: Use of RD in B type.

zeeshanrafique23, 2021-09-22:
https://github.com/lowRISC/ibex/blob/ac8934459b028220ce9fd20683bb018244b59756/rtl/ibex_tracer.sv#L493

I am wondering why there is a use of RD in B-type instructions? Is this an issue of I am missing something.

imphil, 2021-09-23:
Good question, and I can't think of a reason why RD is added here at the moment. *But* I know that I carefully double-checked the output of the tracer against objdump and ensure the two are identical. Can you try to run a binary with a B-type instruction (any branch) on Ibex and see what you get in the tracer, and compare that what you get with objdump?

zeeshanrafique23, 2021-09-23:
I think that's not a problem, that should be same. But there is a column named `Register and memory content` in the trace generated log file which shows content and the variable `data_accessed` is used in that column only.

On this particular instruction `bge x27, x26, 1003b0`, I am getting this output in the log file `x27:0x001004f0 x26:0x001004e4  x0=0x00000000`. Here we have x0=0x00000000 (although we don't need this at all) that because branch not used `rd` so it will not be enabled in `ibex_core.sv`.
https://github.com/lowRISC/ibex/blob/ac8934459b028220ce9fd20683bb018244b59756/rtl/ibex_core.sv#L1356

zeeshanrafique23, 2021-09-23:
Sending PR to fix this issue.

closed 2021-10-01 by zeeshanrafique23