lowRISC/ibex #1444: Building dependency in Fusesoc step fails on Ubuntu 20.04

SBIIT, 2021-09-24:
Hii, I am trying to run the ibex core on my Ubuntu 20.04 system as suggested in https://github.com/lowRISC/ibex/tree/master/examples/simple_system. 

The dependencies are correctly installed, but the fusesoc building simulation step fails inspite of that:
fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast

The error is as follows:

INFO: Running pre_build script check_tool_requirements
INFO: Tool verilator present: Sufficiently recent version (found 4.213; needed 4.028)
INFO: Tool edalize present: Sufficiently recent version (found 0.2.2; needed 0.2.0)
INFO: Building simulation model
verilator -f lowrisc_ibex_ibex_simple_system_0.vc --trace --trace-fst --trace-structs --trace-params --trace-max-array 1024 -CFLAGS "-std=c++11 -Wall -DVM_TRACE_FMT_FST -DTOPLEVEL_NAME=ibex_simple_system -g" -LDFLAGS "-pthread -lutil -lelf" -Wall -Wwarn-IMPERFECTSCH --unroll-count 72
%Warning-UNUSED: ../src/lowrisc_ibex_ibex_top_tracing_0.1/rtl/ibex_top_tracing.sv:27:26: Parameter is not used: 'RndCnstLfsrSeed'
                                                                                       : ... In instance ibex_simple_system.u_top
   27 |   parameter lfsr_seed_t  RndCnstLfsrSeed  = RndCnstLfsrSeedDefault,
      |                          ^~~~~~~~~~~~~~~
                 ... For warning description see https://verilator.org/warn/UNUSED?v=4.213
                 ... Use "/* verilator lint_off UNUSED */" and lint_on around source to disable this message.
%Warning-UNUSED: ../src/lowrisc_ibex_ibex_top_tracing_0.1/rtl/ibex_top_tracing.sv:28:26: Parameter is not used: 'RndCnstLfsrPerm'
                                                                                       : ... In instance ibex_simple_system.u_top
   28 |   parameter lfsr_perm_t  RndCnstLfsrPerm  = RndCnstLfsrPermDefault,
      |                          ^~~~~~~~~~~~~~~
%Error: Exiting due to 2 warning(s)
Makefile:16: recipe for target 'Vibex_simple_system.mk' failed
make: *** [Vibex_simple_system.mk] Error 1
ERROR: Failed to build lowrisc:ibex:ibex_simple_system:0 : 'make' exited with an error code

I am new to ibex, can anyone be kind to help me with the resolution of the error. 
Thanks in advance.

## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

zeeshanrafique23, 2021-09-25:
Seems like these two parameters are declared in the module but not used (IDK why it is), there are two ways to off these warnings.

1. Comment line 27 and 28 in the module `ibex_top_tracing.sv`. https://github.com/lowRISC/ibex/blob/45f727dfc6e59119347aca62e38e69175f680bf6/rtl/ibex_top_tracing.sv#L27 
2. Add `- "-Wno-UNUSED"` after line 146 in [ibex_simple_system.core](https://github.com/lowRISC/ibex/blob/master/examples/simple_system/ibex_simple_system.core) file.

But the first one is recommended.

rswarbrick, 2021-09-27:
These parameters are used when the `SecureIbex` parameter is set, but we forgot to waive them if not: I'll push up a patch in a minute to fix that, thanks. They are to allow systems that integrate the chip to specify the initial state of the LFSR that we use to insert dummy instructions.

rswarbrick, 2021-09-27:
Oh, actually we forgot to connect them at all. Oops!

closed 2021-09-27 by rswarbrick

SBIIT, 2021-09-27:
Wow, Thank you, guys. I just tried the updated code, and everything works perfectly. 
Thanks for promptly solving the issue.

PR #1446 (2021-09-27): Add missing parameters to ibex_top_tracing
Fixes #1444