lowRISC/ibex #1445: Showing warnings related to core file while running simulation

zeeshanrafique23, 2021-09-25:
## Observed Behavior
The warning messages can be seen in Yellow color.  

[image]

## Expected Behavior
TADA! Now they are gone.  

[image]

## Steps to reproduce the issue
I didn't make any modifications to the source code, before running the command. But to resolve the issue I did.

## My Environment
Fusesoc 1.11

**EDA tool and version:**
Tool verilator present: Sufficiently recent version (found 4.212; needed 4.028)
Tool edalize present: Sufficiently recent version (found 0.2.2; needed 0.2.0)

**Operating system:**
Ubuntu 20.04

**Version of the Ibex source code:**

45f727dfc6e59119347aca62e38e69175f680bf6

zeeshanrafique23, 2021-09-25:
Seems like the bug origin files are the IP of opentitan, so send the PR there to fix this issue. https://github.com/lowRISC/opentitan/pull/8372

For ibex, I changed the file and am ready to send the PR, but I am afraid that are you using their IPs as it is (align with their commit hash) or you can make changes. I changes are allowed then I would love to contribute to ibex.

rswarbrick, 2021-09-27:
You're right: we vendor this file from OpenTitan. I'm not sure exactly how the JasperGold flow works there, but I see that you've tagged @cindychip on the PR so she can probably help. The right way to fix this here is to get that merged and then re-vendor the files. Should be simple enough but, in case we forget, please feel free to ping us again on this issue once the OpenTitan PR is merged.

zeeshanrafique23, 2021-09-27:
Great thanks @rswarbrick, I will.

cindychip, 2021-09-27:
Thank you @zeeshanrafique23 and Rupert for bringing up this issue. I think PR #8372 would eliminate this warning. But use the `alert_esc_rxtx_fpv.core` as an example, we wanted to add the `formal` switch so we can only include this core file when the upstream target is formal. If changing it back to `default_tool`, it will be included in all other targets such as design, dv, lint, etc.
I believe Michael helped file an issue in FuseSOC repo.

thanks,
Cindy

zeeshanrafique23, 2021-09-27:
Hi @cindychip,
Thanks for reviewing, seems like the target `formal` is already added in the core file. Changing it to `default_tool` will just tell the fusesoc to make the `icarus` as a default tool when running the `formal` as a target, there are only two targets present in the `prim_esc_rxtx_fvp.core` i.e., `formal` and `lint`.

Now, while comparing both files [prim_esc_rxtx_fvp.core](https://github.com/lowRISC/opentitan/blob/master/hw/ip/prim/fpv/prim_esc_rxtx_fpv.core) and [prim_alert_rxtx_fvp.core](https://github.com/lowRISC/opentitan/blob/master/hw/ip/prim/fpv/prim_alert_rxtx_fpv.core) you will find the target sections same (except for the top module name).

And I can't find the `alert_esc_rxtx_fpv.core`, could you point me to that file please.

cindychip, 2021-09-27:
Hi @zeeshanrafique23 Sorry I made a typo, it should be `prim_esc_rxtx_fvp.core`. 
Let me give it a try locally to double check if the change will affect FPV runs :)

zeeshanrafique23, 2021-10-01:
Hi @rswarbrick, the PR https://github.com/lowRISC/opentitan/pull/8372 is merged.

rswarbrick, 2021-10-05:
Oh, drat. Looks like something has come unstuck with the vendoring update. I won't have time today, but will try to sort things out tomorrow.

closed 2021-11-16 by rswarbrick