lowRISC/ibex #1447: Running a cryptographic library on ibex

SBIIT, 2021-09-28:
Hi, Thanks to the community for solving my earlier issue. 
A little update from my side, the repo now works perfectly well on Ubuntu 20.04 system with Verilaor 4.213. 

This time I have a question regarding running and linking a C library on ibex simple system 
and eventually running it on an FPGA.
My goal is to run a cryptographic code run on the ibex RISC-V core. I am familiar with the mbedTLS or the OpenSSL 
libraries, but how does the integration work for the ibex cores? My question is particularly related to how should I link 
a library and the respective headers on the ibex simple systems example. 

Any lead on this is very very welcome. Has anyone done this before, or seen such integration in a previous repo?
Any input on how to integrate a library would help, it does not have to necessarily be a crypto library. 
Thanks in advance. 

## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

rswarbrick, 2021-09-29:
The basic answer is to build whatever software you're building with an appropriately configured RISC-V toolchain (probably easiest to use the one we provide) and then do the final link with the linker script in `sw/simple_system/common`.

If you're trying to figure out the right incantations for your Makefile, it might help to run `make -C examples/sw/simple_system/hello_test`. The output shows the GCC, objcopy and srec_cat commands that the build flow runs.

rswarbrick, 2021-09-29:
Oh! And this is a [freestanding](https://en.cppreference.com/w/cpp/freestanding) implementation (you'll see `-nostdlib -nostartfiles -ffreestanding` in the compilation lines): it doesn't come with things like IO or dynamic memory management.

SBIIT, 2021-10-01:
Thank you for your support and suggestions. 
Following your earlier comment, I could manage to incorporate a simple header file in the common directory in the simple system example. But when I am trying to incorporate a bigger library, the compilation runs, 
but eventually, the make fails with undefined reference links to standard memory functions. 

The typical error looks like --> "undefined reference to `memcpy'", and many more such as memset, calloc, free, memcmp, printf, fopen etc.etc.
Is this the same thing that you were talking about in your earlier comment "it doesn't come with things like IO or dynamic memory management." ? 
And how to go around the problem? Is there any way that we could include these libraries so that the underfined reference gets solved? Or do you advise avoiding using such functions? But in that case, most of the standard libraries in C would be using references to such functions, they cannot be ported to run on ibex core. 

I am really keen to learn and make this work on the Ibex core. Any input on this or tricks to go about this problem is really really welcome. Thanks in advance.

SBIIT, 2021-10-01:
Also, there is one more issue. After I removed the -nostdlib flag from the common.mk, I found the following error. 

/lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/bin/ld: hello_test.elf section `.text' will not fit in region `ram'
/lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/bin/ld: section .stack VMA [0000000000130000,0000000000131fff] overlaps section .text VMA [0000000000100084,0000000000138967]
/lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/bin/ld: region `ram' overflowed by 193364 bytes

Later, digging deeper found that this error was always there, did not recognize it earlier behind several undefined reference errors.
Any ideas on how this could be solved?

rswarbrick, 2021-10-04:
Hi there! Firstly, you're right that there aren't things like `memcpy` by default. That's why I mentioned the fact that it's a freestanding implementation. As you found, you *can* include more of the standard library by dropping the `nostdlib` flag. That should give you things like `memcpy` but you still won't have a working heap, I don't think (so e.g. `malloc` / `free` won't work)

The link time errors here are just saying that your application is too big to fit in the available memory. You could always hack the simple system linker script (`examples/sw/simple_system/common/link.ld`) to give you more space. For example, I *think* things should work if you increase the length of `ram` and the address of `stack` like this:
```
MEMORY
{
    ram         : ORIGIN = 0x00100000, LENGTH = 0x80000 /* 512 kB */
    stack       : ORIGIN = 0x00180000, LENGTH = 0x8000  /* 32 kB */
}
```

The Verilator-based simulation (top-level at `ibex_simple_system.sv`) allows up to 1MiB of RAM, so I think it should still work. Obviously, this may or may not fit on your chosen FPGA. It definitely won't fit on the Artya A7-35 (see the comment in that file for where the numbers came from before).

closed 2021-12-10 by GregAC