lowRISC/ibex #1466: How to add DDR memory instead of BRAM?

Kian75, 2021-11-03:
Hi I have been working on Ibex recently with Nexys4 DDR FPGA which has 128MiB ddr2 memory... My question is how can I change the current memory of Ibex (which is BRAM of FPGA) to this ddr2 memory? I have a application which needs lots of memory...

GregAC, 2021-11-05:
Hi @Kian75 

You could write an interface adapter between the Ibex top-level memory IO (described here: https://ibex-core.readthedocs.io/en/latest/03_reference/load_store_unit.html#lsu-protocol) and whatever interface is available for the DDR (via a Xilinx controller or a 3rd party one, AXI is often used), though using it directly like that will be very slow.

To get reasonable performance you'll need a cache. Ibex has an optional instruction cache but no data cache, so you'd need to build one.

Kian75, 2021-11-05:
Hi @GregAC 

Thank you for your response... So if I understood correctly you mean I can do this for example: In Vivado there is an IP called Memory Interface Generator which is used for interfacing DDR2 memory that my board has... so I should come up with an interface adapter between Ibex and this interface (which utilize axi as you mentioned most of them do)?

And for cache part... Do you have any link or reference I can start with? Sorry If I am asking low level questions...

GregAC, 2021-11-11:
> so I should come up with an interface adapter between Ibex and this interface (which utilize axi as you mentioned most of them do)?

Yes

> And for cache part... Do you have any link or reference I can start with?

Not really I'm afraid, the wikipedia page may be a reasonable place to start for an overview: https://en.wikipedia.org/wiki/CPU_cache, Googling for 'CPU Cache Verilog' shows up a few things too. You could perhaps look at the data cache in the CVA6 processor here: https://github.com/pulp-platform/cva6/tree/master/src/cache_subsystem though likely a fair amount of work to get it working with Ibex.

Kian75, 2021-11-15:
Thank you for your answer @GregAC ...
By the way for populating this ddr2 memory with .mem files like compiled version of your example here (but I have a much bigger code) [https://github.com/lowRISC/ibex/blob/master/examples/sw/led/led.c](https://github.com/lowRISC/ibex/blob/master/examples/sw/led/led.c) I should use bootloader right?
I searched in many websites and they gave me this answer [https://stackoverflow.com/questions/69962170/how-to-boot-ddr-memory-of-an-fpga/69963257?noredirect=1#comment123691746_69963257](https://stackoverflow.com/questions/69962170/how-to-boot-ddr-memory-of-an-fpga/69963257?noredirect=1#comment123691746_69963257)
So is there any bootloader available for Ibex core?

GregAC, 2021-11-15:
Yes you'll need some kind of loader. You'll need to build one yourself as we don't supply one.

closed 2021-12-10 by GregAC