lowRISC/ibex #1471: [bug] PMP checking incorrect with icache

tomroberts-lowrisc, 2021-11-16:
Since the PMP checking in Ibex is performed on external requests, any changes to PMP registers or changes between U and M mode will render the icache contents out of date. The checking cannot easily be moved after the icache as the address can be misaligned at that point due to compressed instructions.
The following solutions are proposed:
1. Flush the icache (and all buffered instructions) any time a PMP register is written (which should be rare).
2. Store the associated U/M mode status with each cache line so that the icache does not need to be flushed on every mode change (which happens on interrupts and so is more frequent)

GregAC, 2021-11-16:
I'm wondering if we can move it after the icache by assuming the majority of accesses won't cross a PMP boundary and doing a stall where they do.

We'd need some extra logic to say 'this access is next to the PMP region boundary end'. Overlapping PMP regions also complicates matters but perhaps something to say 'this access is next to the PMP region boundary beginning' can deal with that.

So when you get an unaligned access if the first half passes PMP and it's not next to a PMP region boundary end and not next to a PMP region boundary start you know the second half of it will also pass. Otherwise you stall to do two PMP checks.

tomroberts-lowrisc, 2021-11-16:
That might be doable yes. It would be an extra two comparators (plus some adders etc.) per PMP region since those two conditions are not something we already match for, but that might be ok for timing since the address itself is straight out of a flop. We could potentially even just add an extra PMP channel and always check the address plus2 in parallel with the address!?

GregAC, 2021-11-16:
I was hoping the extra logic could be done without an replicated PMP channel, but I have yet to think particularly hard about it!

Replicated channel may be the way to go as it doesn't have adverse timing impact as you say, though does add area obviously.

tomroberts-lowrisc, 2021-11-17:
I've knocked together the replicated channel - looks like ~5.5kGE extra on our synthesis flow so could be a goer. I'll have a look at the reduced option now.

closed 2021-12-10 by GregAC