lowRISC/ibex #1472: [bug, dbg] Don't cache instruction whilst in debug mode

GregAC, 2021-11-17:
The debug module used in conjunction with Ibex implements its functionality by having Ibex execute code that dynamically changes depending upon the commands issued to it. When the icache is present caching the memory region used for this will break the debug module.

A simple fix is to disable instruction caching whilst Ibex is in debug mode.

johngt, 2022-07-20:
Per the comment in the related thread - https://github.com/lowRISC/opentitan/issues/11447 it seems that @GregAC did create a patch for this that may have gone in since D2 and up until now, but it is unclear if this has been merged in.
If it has successfully been merged in, please comment and we can close this issue and the related OT issue.

jamesrbailey, 2022-08-24:
We also encountered this issue. 

As an alternative to disabling icache as suggested here, we also found adding a fence.i instruction to the debug ROM was a sufficient fix to let us leave icache enabled (although we have not rigorously tested this).

PR #1813 (2022-09-27): [rtl] Don't cache instructions in debug mode
RISC-V debug modules may utilise dynamically changing code. Don't cache any instructions in debug mode to correctly support this.

Fixes #1472
commit 7b1be335 (2022-09-27): [rtl] Don't cache instructions in debug mode

RISC-V debug modules may utilise dynamically changing code. Don't cache
any instructions in debug mode to correctly support this.

Fixes #1472
closed 2022-09-27 by GregAC