lowRISC/ibex #1481: [ci] Let's try to integrate a Yosys synthesis flow into CI

rswarbrick, 2021-12-02:
The guys at Antmicro have been working on their Surelog+UHDM+Yosys flow and it seems that it can now build Ibex with vanilla Yosys plus their stuff as a plugin.

It would be neat if we could try running this in CI. We'd get early warning if we made a change that broke their frontend (possibly a sign of a genuine Verilog error) and we'd also get instant kGE estimates.

We'll need to decide which version of the software to track and how to "install" it. Maybe the right approach would be for Antmicro to generate a docker image containing the tool, so we could consume that?

We'll also need to figure out how to run stuff. A GitHub action? Some sort of Azure integration?

rswarbrick, 2021-12-02:
@tgorochowik: Please correct me if I've mangled something here! :-)

tgorochowik, 2021-12-02:
Sure, thanks! If you want to go with the docker image approach we could create an action (similar to the action with verible) and use it. 
This way it should be possible to control the version of yosys/uhdm plugin we're using.
We're of course open to doing it differently too (be it compiling everything in the CI or providing binaries in some way). Thanks again!

johngt, 2022-07-20:
I believe that this effort is still ongoing but will likely not make it before V2, so bringing it into the Project to be tracked as a future effort.

GregAC, 2022-07-20:
Removed from DV project as this is not DV related