lowRISC/ibex #1485: Wrong SV code - compilation error occurs

dariuszst, 2021-12-06:
## Observed Behavior

After https://github.com/lowRISC/ibex/commit/169785d0711335c94561a93146e069766eec138c commit the "ibex_simple_system" and UVM tests cannot be compiled because of the error:

`ALOG: Error: VCP2675 /ibex/build/lowrisc_ibex_ibex_simple_system_0/src/lowrisc_ibex_ibex_top_0.1/rtl/ibex_register_file_fpga.sv : (63, 1): Cannot write to a variable 'mem[k]' that is also driven by an always_comb/always_latch/always_ff procedural block. `

The problem is caused by wrong SV code:
https://github.com/lowRISC/ibex/blob/c78acac8cc381293900f3648e365ddd136784789/rtl/ibex_register_file_fpga.sv#L54-L65

In this case the mem[] variable is written in always_ff  and initial block what is prohibited by SV LRM:

> "The always_ff procedure imposes the restriction that it contains one and only one event control and no
> blocking timing controls. Variables on the left-hand side of assignments within an always_ff procedure,
> including variables from the contents of a called function, shall not be written to by any other process."

## Expected Behavior

The code should compile without errors.

## Steps to reproduce the issue

## My Environment

**EDA tool and version:**

Aldec Riviera-PRO

**Operating system:**

Linux, Ubuntu 20.04

**Version of the Ibex source code:**

rswarbrick, 2021-12-06:
Since the SV spec is rather more free and easy with just `always @(posedge ...)`, maybe we should switch this file to that? (With a comment explaining why we're not following our style guide here).

vogelpi, 2021-12-06:
Ah, yes that could work @rswarbrick . However, also need to ensure the code still infers LUT-RAM on FPGA.

@dariuszst could you please check if switching the `always_ff` to `always @(posedge` here would solve the tooling error you see? I would then have a look at the FPGA synthesis.

dariuszst, 2021-12-07:
OK. I've checked the version with always_ff()  replaced with always(). It seems to work OK for RTL compilation and simulation.

vogelpi, 2021-12-09:
Thanks @dariuszst  for checking this out. I've now filed a PR to fix this.

closed 2021-12-09 by vogelpi