lowRISC/ibex #1489: [icache,dv] Minor tidyups after dropping PMP interface

rswarbrick, 2021-12-10:
Tom's RTL changes at https://github.com/lowRISC/ibex/pull/1474 simplify things dramatically on the DV side, which is nice. I made some suggested simplifications on that PR and there are probably further simplifications possible. Do them!

marnovandermaas, 2022-07-21:
@johngt As far as I can tell the simplifications are mostly done by @GregAC in: https://github.com/lowRISC/ibex/commit/72a892d62ce820d9519825e308df45c05cfbe867
The only other thing Rupert mentions is the:
> need to work on this code to teach it about scrambling

I'm not sure what scrambling means in this case. Besides that I think this issue can be closed.

marnovandermaas, 2022-08-09:
@GregAC is it ok if I close this issue?

marnovandermaas, 2022-08-12:
> The nits mentioned by @rswarbrick are still there so this is still live.

Which nits do you mean? I'm not sure what still needed beyond this commit you've already made: https://github.com/lowRISC/ibex/commit/72a892d62ce820d9519825e308df45c05cfbe867