lowRISC/ibex #1496: riscv_nested_interrupt_test case failed

cnan123, 2021-12-21:
I use the comand ```make TEST=riscv_nested_interrupt_test SEED=8037``` will have one case failed.
Due to random delay for load/store, 750 cycles for timeout sometimes is not enough. when I modify dv/uvm/core_ibex/tests/core_ibex_test_lib.sv line 247 to "check_next_core_status(HANDLING_IRQ, "Core did not jump to vectored interrupt handle", 760) ", then it will pass.

"Synopsys VCS 2020.03-SP1" 
"Ubuntu Linux 18.04".
commit id: fd1195a1

vogelpi, 2021-12-21:
Thanks @cnan123 for reporting this issue. I didn't manage to reproduce this with Synopsys VCS R-2020.12-SP2. Unfortunately, I currently don't have access to 2020.03-SP1 anymore. Maybe someone else from the team with access to this specific version can give it a try. I will bring this up after New Year and report back.

GregAC, 2022-01-06:
Thanks for the report @cnan123. Ultimately this is a similar style of issue to https://github.com/lowRISC/ibex/issues/1499 and the resolution is the same. Our co-simulation work will remove the problem.

johngt, 2022-07-20:
Tagging for closure as we believe that with the new co-sim approach this should be resolved.