lowRISC/ibex #1500: Building Arty A7 A35-T example

kamilrakoczy, 2022-01-03:
## My Environment

**EDA tool and version:**
Xilinx Vivado 2019.2 (64-bit)

**Operating system:**
Arch linux

**Version of the Ibex source code:**
410ffd349da1d4c13f77b208bb3b93adb154fac0 - no changes to the code

I've tried to build Arty A7 example for A35-T using current master, but it looks like there aren't enough resources available on the board. Run command:
```
fusesoc --cores-root=. run --build --tool vivado --target=synth lowrisc:ibex:top_artya7 --part xc7a35ticsg324-1L --SRAMInitFile=`pwd`/../led.vmem
```
Vivado error message:
```
ERROR: [DRC UTLZ-1] Resource utilization: RAMB18 and RAMB36/FIFO over-utilized in Top Level Design (This design requires more RAMB18 and RAMB36/FIFO cells than are available in the target device. This design requires 128 of such cell types but only 100 compatible sites are availabl
e in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)
ERROR: [DRC UTLZ-1] Resource utilization: RAMB36/FIFO over-utilized in Top Level Design (This design requires more RAMB36/FIFO cells than are available in the target device. This design requires 64 of such cell types but only 50 compatible sites are available in the target device. 
Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)
ERROR: [DRC UTLZ-1] Resource utilization: RAMB36E1 over-utilized in Top Level Design (This design requires more RAMB36E1 cells than are available in the target device. This design requires 64 of such cell types but only 50 compatible sites are available in the target device. Please
 analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)
``` 
Targeting A100-T generates bitstream, but description of the ``top_artya7.core`` file suggests it should also support A35-T: https://github.com/lowRISC/ibex/blob/master/examples/fpga/artya7/top_artya7.core#L6.

vogelpi, 2022-01-03:
Hi @kamilrakoczy ,

thanks for your interest in Ibex. Indeed, the current configuration is too big for the A35-T. We have been discussing to re-enable support for it a while back but this will break our CoreMark power analysis setup for Vivado. That's why haven't merged the fix yet. In the meantime, you can cherry-pick the following commit:
https://github.com/lowRISC/ibex/pull/1433/commits/4f876df8e85dda4e117dc576d47b8e6a1a37dcdc 

Or just copy over the change, it's very simple. In `top_artya7.sv`, change:
```sv
 parameter int          MEM_SIZE     = 256 * 1024; // 256 kB
```
to
```sv
  parameter int          MEM_SIZE     = 64 * 1024; // 64 kB
```
This should fix your problem for now.

kamilrakoczy, 2022-01-03:
Hi @vogelpi, thank you for quick answer.

Yes, indeed changing ``MEM_SIZE`` parameter fixed above problem, but now I found another one:
bitstream generated using current master (410ffd3 - only adapting MEM_SIZE to ``64 *1024``) doesn't work on HW. LEDs doesn't blink.

Last working version for me (with blinking LEDs) is: fccdd63992991c65458c94f330a1c7428b7ee6db.

vogelpi, 2022-01-03:
Sorry to hear it still doesn't work. Unfortunately, I don't have an Arty board here to debug. I've now triggered a Vivado build to see if there are some warnings printed during the build.

Do you get any feedback from the tool when trying to load the bitstream and/or load the binary?

kamilrakoczy, 2022-01-04:
I don't get any feedback from the tool when I'm trying to load the bitstream.

I also compared Vivado logs between working and not working versions, but I don't see any new warnings in the not working version.

kamilrakoczy, 2022-01-05:
I've tested A100-T variant and it seems that it also doesn't work using 410ffd3 (without any changes).
Once again last working version for A100-T is from fccdd63.

Looks like it is bug in Ibex code or something changed that was not reflected in fusesoc core files?

GregAC, 2022-01-06:
Thanks for reporting this @kamilrakoczy I do have access to an Arty A7 board with a A35 so I will look into it. It may be a few days til I have the time to look at it though.

GregAC, 2022-01-13:
I believe I have a fix for both issues raised here, it's been added to this PR: https://github.com/lowRISC/ibex/pull/1433

kamilrakoczy, 2022-01-14:
Thanks for working on this @GregAC.

I can confirm that changes in #1433 fixes my problem and bitstream is working again.

commit e53b0339 (2022-01-14): [examples/fpga] Fix memory interface

Logic driving instr_gnt/data_gnt violated Ibex memory protocol. It just
happened to work until a recent change.

Fixes #1500
closed 2022-01-14 by GregAC