lowRISC/ibex #1504: Did you plan to used force-riscv to generate random-instruction in Ibex's dv env?

MasterJerryZh, 2022-01-08:
Hi, as I known, force-riscv can generate random-instruction more faster than riscv-dv because of its base on C++. Did you have a plan to used force-riscv to generate random-instruction in your dv env next up?  It can coexist with riscv-dv. We can only replace the part of instruction generator.

GregAC, 2022-01-08:
I hadn't come across force-riscv before, just had a brief look and it does look like a capable instruction generator.

We have no plans to move away from riscv-dv, in particular speed of instruction generation isn't much of a concern, riscv-dv is fast enough for our purposes and supports what we require for Ibex. However force-riscv has the advantage of not requiring a full system verilog simulator to run, allowing its use with a verilator based testbench. So it may be of interest to us in the future.

Our current DV environment isn't intimately tied to riscv-dv, so using force-riscv in its place is certainly possible though not entirely trivial.

MasterJerryZh, 2022-01-08:
Ok. I have got it. Thanks!

closed 2022-01-08 by MasterJerryZh