lowRISC/ibex #1517: user manual - minor error in the rv32b description

ftorres16, 2022-01-27:
## Observed Behavior

In the [User Manual, section Core Integration, subsection Parameters](https://ibex-core.readthedocs.io/en/latest/02_user/integration.html#parameters), the description for the RV32B parameter says the following:

> “ibex_pkg::RV32Full”: All sub-extensions

But `RV32Full` is not defined in `ibex_pkg.sv`.

```systemverilog
  typedef enum integer {
    RV32BNone       = 0,
    RV32BBalanced   = 1,
    RV32BOTEarlGrey = 2,
    RV32BFull       = 3
  } rv32b_e;
```

## Expected Behavior

The table should say:

> “ibex_pkg::RV32BFull”: All sub-extensions

## Steps to reproduce the issue

Read the manual in: https://ibex-core.readthedocs.io/en/latest/02_user/integration.html#parameters

## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

latest

rswarbrick, 2022-01-27:
Thanks for the bug report: looks like @vogelpi and I fixed it simultaneously! One of the fixes should go in soon :-)

ftorres16, 2022-02-01:
If I go to the Ibex documentation the whole table in the Parameters section is gone now. 

https://ibex-core.readthedocs.io/en/latest/02_user/integration.html#parameters

I tried two different browsers and clearing the cache but no luck so far.

@rswarbrick is it possible that the documentation wasn't built correctly after this?

rswarbrick, 2022-02-02:
Oof. You're completely right. Fixed in #1523.

PR #1518 (2022-01-27): [doc] Fix RV32B enum description
Fixes #1517.