lowRISC/ibex #1521: Tests failing with Timeout error

jayasrikiran, 2022-01-28:
Timeout[10s]: /programs/riscv-gcc-10.2.0-gb693c14--200904T1226/bin/spike --log-commits --isa=rv32imc -l out/seed-123/instr_gen/asm_test/riscv_machine_mode_rand_test_0.o &> out/seed-123/instr_gen/spike_sim/riscv_machine_mode_rand_test.0.log

## My Environment
make TEST=riscv_machine_mode_rand_test ITERATIONS=1 SEED=123 WAVES=1 SIMULATOR=xlm COSIM=0 IBEX_CONFIG=small ISS=spike

**EDA tool and version:**
Using Cadence 20.03-s08

**Operating system:**
Centro 7 

**Version of the Ibex source code:**
https://github.com/lowRISC/ibex.git

SerLippo, 2022-02-25:
Check your spike verison, gcc verison and python requirements. 
Make sure you use every tool that lowRISC maintains for ibex.

rswarbrick, 2022-02-25:
I'm really sorry for not noticing this issue when it was first raised. The problem is probably that you're using the wrong version of Spike. The randomised test binary is probably doing something not supported by the ISA that your Spike is modelling, which means the simulation is sitting in a loop on (probably) the fatal exception handler.

We're going to switch on cosimulation support Real Soon Now, which will make this sort of thing much easier to diagnose (it will be harder to compile against the wrong version and then if something *does* go wrong, you should get instant feedback as soon as the simulations diverge).

jayasrikiran, 2022-02-25:
Got the right version of Spike and the issue is fixed. Thank you!

rswarbrick, 2022-02-25:
Great, thanks for letting us know.

closed 2022-02-25 by rswarbrick