lowRISC/ibex #1522: RVFI generate block access error

stevobailey, 2022-01-30:
https://github.com/lowRISC/ibex/blob/46b7e07098f5900b45a37bc45fb8e7b779776b9d/rtl/ibex_core.sv#L1310-L1311

I find that for this generate block, VCS tries to access the signals for all i values. For example, it complains that this line:
https://github.com/lowRISC/ibex/blob/46b7e07098f5900b45a37bc45fb8e7b779776b9d/rtl/ibex_core.sv#L1375
tries to access `rvfi_stage_halt[0-1]`, which is invalid. I was able to fix this by using a regular (non-generate) for loop. This amounts to swapping lines 1310 and 1311, then changing genvar to int:
```
  always_ff @(posedge clk_i or negedge rst_ni) begin
    for (int i = 0;i < RVFI_STAGES; i = i + 1) begin : g_rvfi_stages
```

GregAC, 2022-01-31:
Was it a warning or an error that you saw? We do use VCS and build/run this without issue.

The RTL that uses `i-1` is inside the else block of if `if(i == 0)` block so `rvfi_stage_halt[0-1]` is never actually attempted though I can see how limited static analysis would flag this up as a problem.

stevobailey, 2022-02-03:
Thanks, apparently it was a warning that we were promoting to an error. When I leave it as a warning, it compiles just fine:
```
Warning-[SIOB] Select index out of bounds
.../ibex/rtl/ibex_core.sv, 1278
"rvfi_stage_mem_wdata[(0 - 1)]"
  The select index is out of declared bounds : [0:0].
  In module instance : u_ibex_core 
  In module : ibex_core.

```

closed 2022-02-03 by stevobailey