lowRISC/ibex #1525: mismatch reported with "csrwi 0x7c0, 1"

luli2021, 2022-02-04:
The testcase riscv_arithmetic_basic_test is failed. I found it caused by the instruction "csrwi 0x7c0, 1" generated by the function init_custom_csr in ibex_asm_program_gen.sv. Spike takes it as an exception because the address 0x7c0 is unknown. When I delete the function, the testcase is passed. The same situation happens when I use the testcase riscv_machine_mode_rand_test and riscv_arithmetic_basic_test. I don't know what the original purpose of the testcases is. Or did I misuse the verification environment?

## My Environment

I use IBEX DV without changes.

**EDA tool and version:**

Cadence Xcelium 20.09

**Operating system:**

CentOS

**Version of the Ibex source code:**

e53b033962d39a09da574216a8c1813afccdd212

rswarbrick, 2022-02-08:
Hi there! I suspect this means that you're not using the right version of Spike. The 0x7c0 CSR is a custom one that we added. @GregAC: Can you remember what magic incantation we have to do to get this to work?

GregAC, 2022-02-14:
Mismatching on `csrwi 0x7c0, 1` would indicate you're not using the right Spike binary. An easy well to tell is to run `spilke -h` and see if there's Ibex specific configuration options listed (`--secure-ibex`` and ``--icache-en``). Note the `SPIKE_PATH` environment variable points to which spike is being used so you want to check that one is the correct one. If you have a different spike on your main $PATH vs $SPIKE_PATH that could cause issues.

luli2021, 2022-02-15:
The problem is solved. The reason is that I forgot to switch to the right branch when I download Spike. Thank you very much!!

rswarbrick, 2022-02-15:
Great, thanks for letting us know!

closed 2022-02-15 by rswarbrick