lowRISC/ibex #1526: Consider adding srecord as an optional pre-requisite in examples/simple_system

ftorres16, 2022-02-08:
## Observed Behavior

I followed the instructions to build the software in `examples/symple_system` but I noticed that the `vmem` file wasn't generated.

Looking into the logs of the `make -C <...>` command, I found the following error message:

```
srec_cat hello_test.bin -binary -offset 0x0000 -byte-swap 4 -o hello_test.vmem -vmem                                                        
make: srec_cat: Command not found 
```
I searched online and found that I needed to install `srecord`, after which the `vmem` file was successfully generated.

I noticed that `srecord` is not listed in the [Prerequisites](https://github.com/lowRISC/ibex/tree/master/examples/simple_system#prerequisites) section of the README.md, where other needed libraries and software tools are specified.

## Expected Behavior

I would expect an extra line in the [Prerequisites](https://github.com/lowRISC/ibex/tree/master/examples/simple_system#prerequisites) section stating the following:

> \# Prerequisites
> - <...>
> - srecord. On Debian/Ubuntu, install it by running `apt-get install srecord`.

Optionally, it could state is optional by saying something like this:

> - srecord. On Debian/Ubuntu, install it by running `apt-get install srecord`. (Optional, needed for vmem generation)

## Steps to reproduce the issue

Follow the instructions for building a simple system in: https://github.com/lowRISC/ibex/tree/master/examples/simple_system

## My Environment

**EDA tool and version:**

lowrisc-toolchain-gcc-rv32imc-20210412-1

**Operating system:**

Ubuntu 20.04

**Version of the Ibex source code:**

69dfa6f8daf294c766a042c640a5830169e5b914

rswarbrick, 2022-02-08:
Good idea! Could you write a PR?

ftorres16, 2022-02-08:
I created #1527, hopefully it's okay :smile:

closed 2022-03-07 by ftorres16