lowRISC/ibex #1546: Error: cannot find default versions of the ISA extension `b'

hasver, 2022-02-27:
## Observed Behavior
When I run `make SIMULATOR=questa ISS=spike TEST=riscv_arithmetic_basic_test` in dv/uvm/core_ibex, It returns the following: 
`Makefile:162: out/.sim-cfg.mk: No such file or directory
mkdir -p out/
mkdir -p out/seed-9043/.metadata
Sun, 27 Feb 2022 12:17:37 INFO     Creating output directory: out/seed-9043/instr_gen
Sun, 27 Feb 2022 12:17:37 INFO     Processing regression test list : riscv_dv_extension/testlist.yaml, test: riscv_arithmetic_basic_test
Sun, 27 Feb 2022 12:17:37 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:10
Sun, 27 Feb 2022 12:17:37 INFO     Processing simulator setup file : /ies/ahmhas5b/proj/dolphin/ramp/ibex/vendor/google_riscv-dv/yaml/simulator.yaml
Sun, 27 Feb 2022 12:17:37 INFO     Found matching simulator: questa
Sun, 27 Feb 2022 12:17:37 INFO     Building RISC-V instruction generator
Sun, 27 Feb 2022 12:17:52 INFO     Running RISC-V instruction generator
Sun, 27 Feb 2022 12:17:52 INFO     Generating 10 riscv_arithmetic_basic_test
Sun, 27 Feb 2022 12:17:52 INFO     Running riscv_arithmetic_basic_test with 1 batches
Sun, 27 Feb 2022 12:17:52 INFO     Running riscv_arithmetic_basic_test, batch 1/1, test_cnt:10
Sun, 27 Feb 2022 12:18:06 INFO     Creating output directory: out/seed-9043/instr_gen
Sun, 27 Feb 2022 12:18:07 INFO     Processing regression test list : riscv_dv_extension/testlist.yaml, test: riscv_arithmetic_basic_test
Sun, 27 Feb 2022 12:18:07 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:10
Sun, 27 Feb 2022 12:18:07 INFO     Compiling out/seed-9043/instr_gen/asm_test/riscv_arithmetic_basic_test_0.S
Assembler messages:
Error: cannot find default versions of the ISA extension `b'
Traceback (most recent call last):
  File "/ies/ahmhas5b/proj/dolphin/ramp/ibex/vendor/google_riscv-dv/run.py", line 1155, in <module>
    main()
  File "/ies/ahmhas5b/proj/dolphin/ramp/ibex/vendor/google_riscv-dv/run.py", line 1133, in main
    args.gcc_opts, args.debug)
  File "/ies/ahmhas5b/proj/dolphin/ramp/ibex/vendor/google_riscv-dv/run.py", line 446, in gcc_compile
    run_cmd_output(cmd.split(), debug_cmd=debug_cmd)
  File "/ies/ahmhas5b/proj/dolphin/ramp/ibex/vendor/google_riscv-dv/scripts/lib.py", line 195, in run_cmd_output
    raise exc
  File "/ies/ahmhas5b/proj/dolphin/ramp/ibex/vendor/google_riscv-dv/scripts/lib.py", line 192, in run_cmd_output
    output = subprocess.check_output(cmd)
  File "/ies/ahmhas5b/opt/anaconda3/lib/python3.7/subprocess.py", line 376, in check_output
    **kwargs).stdout
  File "/ies/ahmhas5b/opt/anaconda3/lib/python3.7/subprocess.py", line 468, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/ies/ahmhas5b/opt/riscv/bin/riscv64-unknown-elf-gcc', '-static', '-mcmodel=medany', '-fvisibility=hidden', '-nostdlib', '-nostartfiles', 'out/seed-9043/instr_gen/asm_test/riscv_arithmetic_basic_test_0.S', '-I/ies/ahmhas5b/proj/dolphin/ramp/ibex/vendor/google_riscv-dv/user_extension', '-T/ies/ahmhas5b/proj/dolphin/ramp/ibex/vendor/google_riscv-dv/scripts/link.ld', '-mno-strict-align', '-o', 'out/seed-9043/instr_gen/asm_test/riscv_arithmetic_basic_test_0.o', '-march=rv32imcb', '-mabi=ilp32']' returned non-zero exit status 1.
make: *** [out/seed-9043/.metadata/instr_gen.compile.stamp] Error 1
`

rswarbrick, 2022-02-28:
Hi there! This looks a lot like you're using a standard RISC-V toolchain without support for the B extension. Have you installed the a "lowRISC toolchain", as described [here](https://ibex-core.readthedocs.io/en/latest/03_reference/verification.html#prerequisites-environment-setup)?

vogelpi, 2022-05-10:
I think Rupert's reply answered the question in this issue and I am thus closing it.

closed 2022-05-10 by vogelpi