lowRISC/ibex #1548: Updating Ibex simple system ISA string

luismarques, 2022-03-04:
The Ibex SS method `SimpleSystem::GetIsaString` currently returns an ISA string that contains `Xbitmanip`. We want to eventually move that to the proper bitmanip subextension names.

rswarbrick, 2022-03-24:
On the simulation side, this needed some Spike support, which is merged in https://github.com/riscv-software-src/riscv-isa-sim/commit/c2f30c33304c96917bdb0ab7e2d4d6a2d71d1d25. Note that this appears in the [ibex-cosim-v0.2](https://github.com/lowRISC/riscv-isa-sim/releases/tag/ibex-cosim-v0.2) tag that we're moving to as a dependency Real Soon Now.

johngt, 2022-07-20:
I believe that Spike support is fully integrated so Ibex SS method should now be correct. @luismarques - could you comment on if you believe this is still an issue or if we can close this?