lowRISC/ibex #1560: Simple system VCS simulation can't find prince_ref.h

ftorres16, 2022-03-14:
## Observed Behavior

When attempting to build and simulate the simple system example with VCS, I am getting errors that it can't find `prince_ref.h`

I get the following error log:

```
[...]
Starting vcs inline pass...                                                                                                                 
                                                                                                                                            
31 modules and 0 UDP read.                                                                                                                  
recompiling package _vcs_DPI_package                                                                                                        
recompiling package ibex_pkg                                                                                                                
recompiling package prim_pkg                                                                                                                
recompiling package prim_cipher_pkg                                                                                                         
recompiling package prim_ram_1p_pkg                                                                                                         
recompiling package prim_ram_2p_pkg                                                                                                         
recompiling package prim_secded_pkg                                                                                                         
recompiling package prim_util_pkg                                                                                                           
recompiling package ibex_tracer_pkg                                                                                                         
recompiling module ibex_tracer                                                                                                              
recompiling module prim_generic_ram_2p                                                                                                      
recompiling module prim_buf                                                                                                                 
recompiling module prim_clock_gating                                                                                                        
recompiling module ibex_alu                                                                                                                 
recompiling module ibex_controller                                                                                                          
recompiling module ibex_cs_registers                                                                                                        
recompiling module ibex_csr                                                                                                                 
recompiling module ibex_counter                                                                                                             
recompiling module ibex_decoder                                                                                                             
recompiling module ibex_ex_block                                                                                                            
recompiling module ibex_id_stage                                                                                                            
recompiling module ibex_if_stage                                                                                                            
recompiling module ibex_multdiv_fast                                                                                                        
recompiling module ibex_wb_stage                                                                                                            
recompiling module ibex_core                                                                                                                
recompiling module ram_2p                                                                                                                   
recompiling module bus                                                                                                                      
recompiling module simulator_ctrl                                                                                                           
recompiling module timer                                                                                                                    
recompiling module ibex_register_file_ff                                                                                                    
recompiling module ibex_simple_system                                                                                                       
All of 31 modules done                                                                                                                      
make[1]: Entering directory '<my working dir>/riscv/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-vcs/csrc'  
make[1]: Leaving directory '<my working dir>/riscv/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-vcs/csrc'  
make[1]: Entering directory '<my working dir>/riscv/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-vcs/csrc' 
rm -f _cuarc*.so _csrc*.so pre_vcsobj_*.so share_vcsobj_*.so
gcc -w  -pipe -fPIC -O -I/CMC/tools/synopsys/vcs_vQ-2020.03/vcs/Q-2020.03/include    -c ../../src/lowrisc_dv_secded_enc_0/secded_enc.c     
../../src/lowrisc_dv_scramble_model_0/scramble_model.cc:14:10: fatal error: prince_ref.h: No such file or directory                        
 #include "prince_ref.h"
          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [filelist:13: scramble_model.d] Error 1
make[1]: *** [Makefile:101: product_clean_order] Error 2
make[1]: Leaving directory '<my working dir>/riscv/ibex/build/lowrisc_ibex_ibex_simple_system_0/sim-vcs/csrc'  
Make exited with status 2
CPU time: 1.940 seconds to compile + .398 seconds to elab + .304 seconds to link          
```

(Note that in this post I replaced the actual absolute path from my system with `<my working dir>`)

This only happens in my system with VCS, not with Verilator.

## Expected Behavior

I expected the build to succeed, as it does when I use verilator with the following command:

```
fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast
```

## Steps to reproduce the issue

I am following the instructions to simulate the simple system according to the [README.md](https://github.com/lowRISC/ibex/blob/master/examples/simple_system/README.md) instructions.

Run the following command from the Ibex root directory:

```
fusesoc --cores-root=. run --target=sim --tool=vcs --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast --SRAMInitFile=./examples/sw/simple_system/hello_test/hello_test.vmem
```

## My Environment

**EDA tool and version:**

- Synopsys VCS Q-2020.03

**Operating system:**

| | | 
| :-- | :-- |
| Operating System: | Debian GNU/Linux 10 (buster) |
| Kernel: | Linux 4.19.0-19-amd64 |
| Architecture: | x86-64 |

**Version of the Ibex source code:**

Latest version from `master` (094451a9489d631895101a5de19a26cd4cf4fd7d)

ctopal, 2022-03-31:
Hello again! 

Commits 4c1a4ed1df4c9b16f9b380c8c507a823cb6a98c0 and 68b56ef0f5c0ba8058e8cb512fe724263ed874f2 solves this issue. Could you please try again with the latest version from master?

Thanks,
Canberk

ftorres16, 2022-03-31:
Hello! This works fine now. Than you for your help. :smile: 

You may close the issue

closed 2022-03-31 by a-will