lowRISC/ibex #1566: Rip out all the non-cosim stuff from dv/uvm/core-ibex scripts and docs

rswarbrick, 2022-03-16:
Depends on #1565 but, once we've got working "cosim tooling", we should get rid of the old way of doing things.

This doesn't include the work to rip out the old sequences for e.g. exceptions: we'll track that separately.

johngt, 2022-06-29:
The dependency [#1565](https://github.com/lowRISC/ibex/issues/1565) has been closed out by Greg on 10 May.
As a tldr; for Cosim tooling:
Spike with cosim support. Main thing discussed in #1565 was to get Spike running with Cosim and then deb package it and have it run on the CI.

Main files that reference this are under:
https://github.com/lowRISC/ibex/tree/master/dv/uvm/core_ibex
Namely; Makefile and compare.py. The tests will also use a post_compare flag that will need to be deprecated/removed.

Estimate of time needed is half a day for documentation removal of any post_compare and non-cosim tooling and a couple of hours for modifying the code to remove references. Total est time: 1 day.

johngt, 2022-07-19:
Adding this related issue that is touched upon by cosim changes.
https://github.com/lowRISC/ibex/issues/1669

johngt, 2022-07-25:
Per the Thursday HW Ibex DV meeting. Splitting this issue into two parts.
This issue has been re-labeled with the tools and scripts elements. New issue related to documentation to be created. This issue should then be able to be closed off by @hcallahan-lowrisc fairly soon. Believe that it is waiting on a PR that @hcallahan-lowrisc can link here.

hcallahan-lowrisc, 2022-07-28:
@johngt Will be closed by #1705, which addresses this among other refactoring changes.

johngt, 2022-08-18:
@hcallahan-lowrisc  - believe that https://github.com/lowRISC/ibex/pull/1705 is now merged so that this can now be closed out. Not tagging is completed just yet.

hcallahan-lowrisc, 2022-08-19:
@johngt I've created one final small PR to remove the last few items. I've setup that PR to close this issue.

PR #1788 (2022-08-19): Remove the final cosim flags from core_ibex uvm environment
We are running with cosim by default now, and no longer support COSIM=0. Hence
this option and all downstream conditional paths are no longer required.

~~This also removes the +disable_cosim flag from all tests, as we are working towards all tests running under cosimulation anyway~~. For the bitmanipulation and csr test, any failures are now no-longer due to cosimulation and should be tracked separately. For bitmanipulation, it appears the tests for the `opentitan` configuration are all passing.

`+disable_cosim=1` has it's behaviour [... cut]
closed 2022-08-19 by hcallahan-lowrisc