lowRISC/ibex #1567: Migrate all remaining test sequences to use cosim

rswarbrick, 2022-03-16:
Figure out what needs to be done with e.g. the self-checking interrupt sequences now that we're using cosimulation. Do it!

rswarbrick, 2022-03-17:
The problem here is that some of the interrupt/exception tests have historically worked by carefully modelling exactly what should happen. This is a bit brittle and is causing sporadic failures.

Now that we've got cosimulation support, lots of this checking is unneeded. Work out how to rip out as much as possible, making the tests more robust.

johngt, 2022-06-30:
Based on discussion with @hcallahan-lowrisc yesterday it sounds like he is best placed for working on this task. Assigning to him and changing status accordingly.

Related to #1566 and #1568

GregAC, 2023-02-27:
This is still live in that various tests still have fragile checking that can be replaced by cosim. For M2.5 only need to rip this out where it's causing regression issues, we don't need to fix this up in general.

GregAC, 2023-03-23:
Moving this out of M2.5 in favour of more focussed issue: https://github.com/lowRISC/ibex/issues/2005