lowRISC/ibex #1569: Add support for passing fixed binaries to our uvm/core_ibex regression flow

rswarbrick, 2022-03-16:


GregAC, 2022-05-10:
This is possible, but awkward. Further work required to make this easy and document.

Two aspects to address:
1. Manual runs, run binary with specified UVM test
2. Regression runs, specify fixed binary for test in regression

rswarbrick, 2022-05-10:
(1) "sort of works", in that there are scripts that you can run by hand. But the user interface is awful (lots of required arguments). We should either teach the existing script to infer more (not necessary when run through Make, but useful on the command line) or add yet another wrapper.

For (2), I think there's some support in riscv-dv for adding directed tests. That would probably be the right place to do it. I suspect all my hacky integration scripts will break! But they shouldn't be too hard to fix, I hope.

johngt, 2022-07-20:
Related issue - https://github.com/lowRISC/ibex/issues/1622. Not a priority for now, have informed original poster that this issue is not on the immediate radar.

GregAC, 2023-02-27:
This is now possible via the new directed test flow

closed 2023-02-27 by GregAC