lowRISC/ibex #1579: Implement ePMP tests

GregAC, 2022-03-18:
RISC-DV only generates code to test normal PMP, this needs to be extended to stimulate ePMP functionality.

haroonshafique-10xe, 2022-03-27:
Hi @GregAC 

This issue is for adding/extending the support for ePMP functionality tests in riscv-dv. Right?
Also, which ISS you are using for ePMP? 
I think mainstream spike does not support ePMP. I know a fork which has ePMP support https://github.com/joxie/riscv-isa-sim.

Thanks.

GregAC, 2022-03-28:
The majority of the work is likely to be in riscv-dv yes.

We're using Spike as our simulator. It is unfortunate ePMP hasn't reached the upstream yet. However as it's fairly simple we should be able to rebase the relevant patches from the fork on upstream or possibly implement from scratch.

haroonshafique-10xe, 2022-03-28:
Right.

> rebase the relevant patches from the fork on upstream

Does it mean that the changes from the fork (joxie/riscv-isa-sim) will be rebased on upstream (riscv-software-src/riscv-isa-sim) and then make a PR from their into the upstream? 

> possibly implement from scratch.

Does it mean to implement in the fork (lowrisc/riscv-isa-sim) only for now?

rswarbrick, 2022-03-28:
Hi there! Yes, we'll probably try to take those changes if possible. We don't want to diverge from upstream, so will do our best to get the changes upstreamed if we can.

haroonshafique-10xe, 2022-03-28:
Hi @rswarbrick 

Yes, sounds good.

I am not sure whether you have seen it or not, there is a [test-suite](https://github.com/joxie/riscv-isa-sim/tree/master/tests/mseccfg) already available in the https://github.com/joxie/riscv-isa-sim.That can be used if in immediate need for basic testing. 

However, I plan to add support for ePMP in the riscv-dv.

vogelpi, 2022-06-09:
Update: We have now a new version of Spike with ePMP support (v.0.3, CI keeps using v.0.2 at the moment) and RISCV-DV with basic ePMP support has been vendored in. To enable ePMP checking, one needs to use Spike v.0.3 and inside `riscv_core_setting.sv` one has to set `support_epmp` to `1`.

marnovandermaas, 2022-07-08:
I've made an initial attempt at enabling ePMP tests here: https://github.com/lowRISC/ibex/pull/1695
It's not done yet because I need to make sure to hit some interesting cases especially for Machine Mode Lockdown.
I've followed the specification that I found here: https://github.com/riscv/riscv-tee/blob/main/Smepmp/Smepmp.pdf

marnovandermaas, 2022-07-08:
Currently both the basic MML and MMWP tests are failing. From the traces, I think that Spike has the correct behavior, while ibex doesn't.
- When MML is enabled and LRWX are all 1, Spike throws and exception on the next instruction because it should be a read only region, while ibex happily continues.
- When MMWP is enabled ibex for some reason changes the `x25` destination and source register to `x0`

My proposal is that I add a few more Smepmp tests to hit interesting cases, and then the behavior divergences I mention above can be put into their own issues.

marnovandermaas, 2022-07-14:
> Currently both the basic MML and MMWP tests are failing. From the traces, I think that Spike has the correct behavior, while ibex doesn't.
> 
>     * When MML is enabled and LRWX are all 1, Spike throws and exception on the next instruction because it should be a read only region, while ibex happily continues.
> 
>     * When MMWP is enabled ibex for some reason changes the `x25` destination and source register to `x0` two instructions after the `csrwi` instruction.
> 
> 
> My proposal is that I add a few more Smepmp tests to hit interesting cases, and then the behavior divergences I mention above can be put into their own issues.

Most of these issues actually had to do with RISCV-DV ignoring the configuration passed through `+pmpregion_0` and `MSECCFG` being set before the PMP regions were properly configured. The pull request is now passing the basic Smepmp tests, but there are still some interesting divergences found by the PMP full random tests.

johngt, 2022-07-20:
Per what @marnovandermaas has said in the thread, this issue is now pending on riscv-dv PR being accepted. Please see related PR at https://github.com/google/riscv-dv/pull/873

marnovandermaas, 2022-07-22:
I updated the RISCV-DV pull request based on @vogelpi input. You can find the update here: https://github.com/google/riscv-dv/pull/873

> spent 2

marnovandermaas, 2022-08-01:
> I updated the RISCV-DV pull request based on @vogelpi input. You can find the update here: [google/riscv-dv#873](https://github.com/google/riscv-dv/pull/873)
> 
> > spent 2

This pull request was merged and I vendored this in. I also made some additional changes suggested by Canberk and did some final testing. I updated the estimate to close this pull request to 2 assuming that the changes are approved.

> spent 4

closed 2022-08-03 by marnovandermaas