lowRISC/ibex #1586: Unique case isn't unique in ibex_id_stage.sv

prdwivedi, 2022-03-28:
## Observed Behavior
Ran into this simulation watning:
```
Warning-[RT-MTOCMUCS] More than one condition match in statement
../lowrisc-ibex/rtl/ibex_id_stage.sv, 788
More than one condition matches are found in 'unique case' statement inside 
ibex_simple_system.u_top.u_ibex_top.u_ibex_core.id_stage_i, at time 240s.
  
  Line number 789 and 799 are overlapping.
```

## Steps to reproduce the issue
You can add the following code to any piece of firmware and commenting out the `nop` instruction between `divu` and `lw` causes the warning. 
```c
/* load immediate: set a5 to fixed value of 10 */
asm ( "li	a5,10" );
asm ("nop");
asm ( "li	a1,1000" );
asm ("nop");

/* divide unsigned: set a1 to the result of a1/a5 using unsigned division */
asm ( "divu	a1,a1,a5" );
 
// asm ( "nop" );
  
/* load word: set a5 to contents of a0 plus 4 */
asm ( "lw	a5,4(a1)" );
```

## My Environment
**Version of the Ibex source code:**
We ported Ibex (5350c13ae3dd5c13dbbb60db8145bec573d0953e) and use it as a submodule for our chip top repository. When writing timing critical functions like interrupt handlers, a divide operation in the firmware doesn't let the handler function return and this warning becomes a bigger problem.

**EDA tool and version:**
Synopsys VCS R-2020.12-SP2-0_Full64

**Operating system:**
Red Hat Enterprise Linux release 8.4 (Ootpa)
LSB version: core-4.1-amd64:core-4.1-noarch

prdwivedi, 2022-03-28:
Similar to #845 (Unique case isn't unique in ibex_alu.sv)

rswarbrick, 2022-03-28:
Hi there! Thanks for the report. In the issue you linked to, we figured out that this is a warning from VCS that we didn't care about (and disabled with the `-xlrm uniq_prior_final` compile switch).

Could you take a look at the waves and confirm whether the cases really do overlap?

(Also, could you give the SHA of the repository that you're using? The line numbers don't match the current tip of the master branch)

prdwivedi, 2022-03-31:
Hi! unfortunately, we don't have our repository on GitHub yet, but this was the problem I had:
I was modifying a timer driver and one of the functions used a division operation, and every time I'd enable the division, my interrupt handler function would never quit. Because of the `unique case` warning, I assumed it was because of the division operation, but turns out the signals don't actually overlap and my problem was caused by the counter threshold being too low for the timer to restart on time. 
[image]
Enabling the `-xlrm uniq_prior_final` compile switch disables the warning and setting a higher threshold gets rid of my problem. 
Thanks a lot for replying and sorry about the confusion.

closed 2022-03-31 by prdwivedi