lowRISC/ibex #1587: changing MEM_SIZE does not affect BRAM utilization

mbaykenar, 2022-03-28:
I wanted to check if 2 KB memory works with IBEX ARTY A7 example. The default is 64 KB with FPGAPowerAnalysis option set to '0' and 256 KB otherwise.

I changed top_artya7.sv file such as:
//  parameter int          MEM_SIZE          = FPGAPowerAnalysis == 0 ? 64 * 1024 : 256 * 1024;
 parameter int          MEM_SIZE          = 2 * 1024; // use 2 KB RAM for instr & data

I checked both synthesis results for 64 KB and 2 KB, somehow BRAM utilization numbers are the same: 16

I checked synth reports (top_artya7.vds) and synth utilization reports (top_artya7_utilization_synth.rpt) and be sure that the parameters related to memory are really consistent according to my changes: 

The log for 2 KB synth from top_artya7.vds

INFO: [Synth 8-6157] synthesizing module 'top_artya7' [/home/mbaykenar/ibex_mba/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_ibex_top_artya7_0.1/rtl/top_artya7.sv:5]
    Parameter FPGAPowerAnalysis bound to: 32'sb00000000000000000000000000000000
    Parameter MEM_SIZE bound to: 32'sb00000000000000000000100000000000
    Parameter MEM_START bound to: 0 - type: integer
    Parameter MEM_MASK bound to: 32'b00000000000000000000011111111111
    Parameter SRAMInitFile bound to: ../../../../../examples/sw/led/led.vmem - type: string

INFO: [Synth 8-6157] synthesizing module 'ram_2p' [/home/mbaykenar/ibex_mba/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_ibex_fpga_xilinx_shared_0/rtl/ram_2p.sv:11]
    Parameter Depth bound to: 32'sb00000000000000000000001000000000
    Parameter MemInitFile bound to: ../../../../../examples/sw/led/led.vmem - type: string
    Parameter Aw bound to: 32'sb00000000000000000000000000001001
INFO: [Synth 8-6157] synthesizing module 'prim_ram_2p' [/home/mbaykenar/ibex_mba/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_prim_abstract_ram_2p_0/prim_ram_2p.sv:14]
    Parameter Width bound to: 32'sb00000000000000000000000000100000
    Parameter Depth bound to: 32'sb00000000000000000000001000000000
    Parameter DataBitsPerMask bound to: 32'sb00000000000000000000000000000001
    Parameter MemInitFile bound to: ../../../../../examples/sw/led/led.vmem - type: string
    Parameter Aw bound to: 32'sb00000000000000000000000000001001
INFO: [Synth 8-6157] synthesizing module 'prim_generic_ram_2p' [/home/mbaykenar/ibex_mba/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_prim_generic_ram_2p_0/rtl/prim_generic_ram_2p.sv:9]
    Parameter Width bound to: 32'sb00000000000000000000000000100000
    Parameter Depth bound to: 32'sb00000000000000000000001000000000
    Parameter DataBitsPerMask bound to: 32'sb00000000000000000000000000000001
    Parameter MemInitFile bound to: ../../../../../examples/sw/led/led.vmem - type: string
    Parameter Aw bound to: 32'sb00000000000000000000000000001001
    Parameter MaskWidth bound to: 32'sb00000000000000000000000000100000

The log for 64 KB synth:

INFO: [Synth 8-6157] synthesizing module 'top_artya7' [/home/mbaykenar/ibex_mba/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_ibex_top_artya7_0.1/rtl/top_artya7.sv:5]
    Parameter FPGAPowerAnalysis bound to: 32'sb00000000000000000000000000000000
    Parameter MEM_SIZE bound to: 32'sb00000000000000010000000000000000
    Parameter MEM_START bound to: 0 - type: integer
    Parameter MEM_MASK bound to: 32'b00000000000000001111111111111111
    Parameter SRAMInitFile bound to: ../../../../../examples/sw/led/led.vmem - type: string

INFO: [Synth 8-6157] synthesizing module 'ram_2p' [/home/mbaykenar/ibex_mba/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_ibex_fpga_xilinx_shared_0/rtl/ram_2p.sv:11]
    Parameter Depth bound to: 32'sb00000000000000000100000000000000
    Parameter MemInitFile bound to: ../../../../../examples/sw/led/led.vmem - type: string
    Parameter Aw bound to: 32'sb00000000000000000000000000001110
INFO: [Synth 8-6157] synthesizing module 'prim_ram_2p' [/home/mbaykenar/ibex_mba/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_prim_abstract_ram_2p_0/prim_ram_2p.sv:14]
    Parameter Width bound to: 32'sb00000000000000000000000000100000
    Parameter Depth bound to: 32'sb00000000000000000100000000000000
    Parameter DataBitsPerMask bound to: 32'sb00000000000000000000000000000001
    Parameter MemInitFile bound to: ../../../../../examples/sw/led/led.vmem - type: string
    Parameter Aw bound to: 32'sb00000000000000000000000000001110
INFO: [Synth 8-6157] synthesizing module 'prim_generic_ram_2p' [/home/mbaykenar/ibex_mba/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_prim_generic_ram_2p_0/rtl/prim_generic_ram_2p.sv:9]
    Parameter Width bound to: 32'sb00000000000000000000000000100000
    Parameter Depth bound to: 32'sb00000000000000000100000000000000
    Parameter DataBitsPerMask bound to: 32'sb00000000000000000000000000000001
    Parameter MemInitFile bound to: ../../../../../examples/sw/led/led.vmem - type: string
    Parameter Aw bound to: 32'sb00000000000000000000000000001110
    Parameter MaskWidth bound to: 32'sb00000000000000000000000000100000

I see that the parameters on prim_generic_ram_2p.sv are elaborated correctly:

2 KB (32-bit * 512 = 2 KB)
    Parameter Width bound to: 32'sb00000000000000000000000000100000
    Parameter Depth bound to: 32'sb00000000000000000000001000000000
64 KB (32-bit * 16384 = 64 KB)
    Parameter Width bound to: 32'sb00000000000000000000000000100000
    Parameter Depth bound to: 32'sb00000000000000000100000000000000

I also checked detailed module level utilization and found out that all the BRAM instances are utilized for ram_2p. The register file utilizes LUT RAMs.

Do I miss a detail on synthesizing for 2 KB RAM or what can cause inferring same amount of BRAM for 64 KB and 2 KB ?

Regards,

## My Environment

**EDA tool and version:**
VIVADO 2020.2

**Operating system:**
UBUNTU 20.04

**Version of the Ibex source code:**
6bb67e20f8a7c670e4baf5fe8f130a5ef360f411

vogelpi, 2022-03-31:
Hi @mbaykenar ,

thanks for filing this issue and the details. It's worth noting that all BRAM instances are 36 bits wide and 1000 entries deep (according to the Xilinx documentation). So assuming that the design uses both BRAM ports the BRAM utilization will always be the same no matter whether an implementation uses 10, 100, or 1000 entries.

Now for this example, the 64 KiB config will uses a depth of 16384 which should result in using 17 instances. However, according to your logs only 16 instances are used. Since the BRAMs are 1000 entries deep, I suspect the tool does something "smart" to distribute the 32-bit words to the 36-bit wide BRAM instances.

I would expect the utilization of the 2 KiB config to be smaller and use a single BRAM instance only (512 entries of 1000 entries used). However, your logs tell a different story. I am not 100% sure but I can imagine the following: If you investigate the ram_2p primitive you will see that it's 1) highly configurable and 2)that the BRAMs are auto-inferred from behavioral code. Getting this right for all possible parameterizations can be challenging. I suspect that due to the byte enable (Ibex can not just write full words but also individual bytes) and how the behavioral code is written, Vivado decides to map the 512 32-bit entries onto 16 BRAMs. I can remember we had issues with that a while ago and that Vivado is extremely picky with the byte enable.

vogelpi, 2022-04-01:
Thanks a lot Burak for having a closer look and for feeding back the result of your investigation. This makes a lot of sense!

I think the parameter is by default 1 because that turned out to be better for some ASIC memories in combination with ECC or parity and byte writes.

Anyway, I've now filed a [PR here](lowRISC/Ibex#1595) to modfiy the `ram_2p` module used by our examples (targeting simulation and FPGA only).

mbaykenar, 2022-04-01:
It is my pleasure to work and investigate ibex core.
One thing, I think the link embedded to "PR here" is not correct. It should be 

https://github.com/lowRISC/ibex/pull/1595

instead of 

https://github.com/lowRISC/ibex/issues/lowRISC/Ibex#1595

vogelpi, 2022-04-01:
Thanks, I've corrected the link ;-)

mbaykenar, 2022-04-04:
I am closing issue with one last small correction:

"_It's worth noting that all BRAM instances are 36 bits wide and 1000 entries deep (according to the Xilinx documentation)._"

This statement is nearly correct but with one minor detail: The depth of the BRAM primitive is 1024 instead of 1000 when 36 bit width is selected. You can find details the Xilinx doc of "7 Series FPGAs Memory Resources" UG473 - Table 1-13 and Table 1-14 Port Aspect Ratio:

https://docs.xilinx.com/v/u/en-US/ug473_7Series_Memory_Resources

"_Now for this example, the 64 KiB config will uses a depth of 16384 which should result in using 17 instances. However, according to your logs only 16 instances are used. Since the BRAMs are 1000 entries deep, I suspect the tool does something "smart" to distribute the 32-bit words to the 36-bit wide BRAM instances._"

So that's the reason with 64 KB MEM_SIZE Vivado infers 16 BRAM instances (RAMB36E1) instead of 17.

Regards,
Burak

closed 2022-04-04 by mbaykenar