lowRISC/ibex #1608: Ask a question related to Ibex

mbaykenar, 2022-04-13:
I want to create more working examples of FPGA based system, since ArtyA7 is not available on our team. 

Zedboard, NexysA7 and Genesys2 are my main focuses for now.

Do you have any plan/idea to extend FPGA examples folder? If yes, I can contribute github repo after finalizing and test these different board based ibex cores.

rswarbrick, 2022-04-13:
@GregAC: I think you know the most about our FPGA plans?

GregAC, 2022-04-13:
We don't have any specific plans for further examples.

We are trying to keep the Ibex repository as Ibex focused as possible so new examples would need to be a similar level of complexity to the existing FPGA example rather than an actual full system. My ibex super system repository (https://github.com/GregAC/ibex_super_system) has a more complete FPGA example.

mbaykenar, 2022-04-13:
ok I will look into this super_system example
I still be implementing ibex for Zedboard, NexysA7 and Genesys2. So anyone needs setup for these boards can interact me

closed 2022-04-13 by mbaykenar