lowRISC/ibex #1619: Log comparison fails between RTL & OVPSIMPLUS even in default setup

kumarbrk9, 2022-04-17:
Hi,
I've cloned the Ibex setup & downloaded OVPSIMPLUS after registering. I was trying to run with default setup (as is) without making any modifications.
Run command used: make SIMULATOR=vcs ISS=ovpsim TEST=riscv_arithmetic_basic_test ITERATIONS=1 TARGET=rv32imc ISA=rv32imc
conversion of log to csv is also performed. But during comparison, it says
[FAILED]: 127 matched, 676313 mismatch
0.00% PASS 0 PASSED, 1 FAILED

When I'm trying to run any different tests, same is the case. Almost at same lines it shows mismatch and count is also mostly similar.

My expectation is it should pass as I did not disturb any setup or any files internally and am running as is.
Can you please help if there is anything that I'm missing out ? or Is it like default can fail unless we make some modifications to any of the files?

Thanks in advance.

vogelpi, 2022-04-18:
Hi @kumarbrk9 ,
thanks for opening this issue. We focus our Ibex DV efforts around the [open-source ISS Spike](https://github.com/riscv-software-src/riscv-isa-sim). In the past, we also used to support OVPsim but IIRC it doesn't natively support our latest co-simulation approach. I would assume that's why you get so many mismatches.

I would recommend to try out Spike instead. [As documented in our Ibex manual](https://ibex-core.readthedocs.io/en/latest/03_reference/verification.html?highlight=verification), you'll need to build it from our fork at the moment. @rswarbrick has been working on upstreaming the changes in our fork to simplify things in the future. But I think we are not entirely there yet.

kumarbrk9, 2022-04-18:
Hi Vogelpi,
Got it. Thanks a lot for your kind response.

closed 2022-04-18 by kumarbrk9