lowRISC/ibex #1622: No Sequencer for Request Agent in the Memory Interface Agents directory.

quswar-abid, 2022-04-20:
## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

rswarbrick, 2022-04-20:
The easiest approach is probably to extend the stuff in `dv/uvm/core_ibex` to support the "please run this binary for me" flow. At the moment, that doesn't exist (it's on the near-term list, tracked as #1569).

As a word of warning, I'm currently hacking on the scripts in that directory. Barring disaster, things should settle down again in a day or two.

johngt, 2022-07-20:
Going to tag this as a PendingClose Issue and then close it off as the associated work is tracked in #1569. As a note to the original poster, these efforts have moved from the near-term tracking items to mid to long term.

quswar-abid, 2022-11-17:
Noted, ping me @ quswar.abid@10xengineers.ai when done and I'll close the issue.

Thanks, :)