lowRISC/ibex #1638: HW breakpoints: "invalid hex 116" message from GDB during Opentitan Verilator simulation

bilgiday, 2022-04-26:
## My Environment

**EDA tool and version:**
Verilator 4.210 2021-07-07 rev v4.210
(openocd) Open On-Chip Debugger 0.11.0
(riscv32-unknown-elf-gdb) GNU gdb (crosstool-NG 1.24.0.498_5075e1f) 11.1

**Operating system:**
PRETTY_NAME="Debian GNU/Linux rodete"
NAME="Debian GNU/Linux"
VERSION_ID="rodete"
VERSION="12 (rodete)"
VERSION_CODENAME=rodete
ID=debian

**Version of the Ibex source code:**
Two versions:
- Version 1: I work on the `OpenTitan` repo v: dd14447ef716bb62ce539e48b4841c7f4e1d2ed7
  - In [[rtl] Add support for additional HW breakpoints #1125](https://github.com/lowRISC/ibex/pull/1125), `DbgHwBreakNum` were added to support multiple hardware breakpoints in the Ibex core. I did the following modifications in my local Opentitan repo to enable 8 breakpoints in the `earlgrey` configuration of `Opentitan`:

  ```
  diff --git a/hw/ip/rv_core_ibex/data/rv_core_ibex.hjson b/hw/ip/rv_core_ibex/data/rv_core_ibex.hjson
  index 0f05f52cf..d82feb8b9 100644
  --- a/hw/ip/rv_core_ibex/data/rv_core_ibex.hjson
  +++ b/hw/ip/rv_core_ibex/data/rv_core_ibex.hjson
  @@ -317,6 +317,13 @@
         expose:  "true"
       },
   
  +    { name:    "DbgHwBreakNum"
  +      type:    "int unsigned"
  +      default: "8"
  +      local:   "false"
  +      expose:  "true"
  +    },
  +
       { name:    "SecureIbex"
         type:    "bit"
         default: "0"
  diff --git a/hw/ip/rv_core_ibex/rtl/rv_core_ibex.sv b/hw/ip/rv_core_ibex/rtl/rv_core_ibex.sv
  index 847168636..56ca88c81 100644
  --- a/hw/ip/rv_core_ibex/rtl/rv_core_ibex.sv
  +++ b/hw/ip/rv_core_ibex/rtl/rv_core_ibex.sv
  @@ -29,6 +29,7 @@ module rv_core_ibex
     parameter bit                   ICacheScramble   = 1'b1,
     parameter bit                   BranchPredictor  = 1'b1,
     parameter bit                   DbgTriggerEn     = 1'b1,
  +  parameter int unsigned          DbgHwBreakNum     = 1,
     parameter bit                   SecureIbex       = 1'b1,
     parameter ibex_pkg::lfsr_seed_t RndCnstLfsrSeed  = ibex_pkg::RndCnstLfsrSeedDefault,
     parameter ibex_pkg::lfsr_perm_t RndCnstLfsrPerm  = ibex_pkg::RndCnstLfsrPermDefault,
  @@ -355,6 +356,7 @@ module rv_core_ibex
       .ICacheScramble           ( ICacheScramble           ),
       .BranchPredictor          ( BranchPredictor          ),
       .DbgTriggerEn             ( DbgTriggerEn             ),
  +    .DbgHwBreakNum            ( DbgHwBreakNum            ),
       // SEC_CM: LOGIC.SHADOW
       // SEC_CM: PC.CTRL_FLOW.CONSISTENCY, CTRL_FLOW.UNPREDICTABLE, CORE.DATA_REG_SW.SCA
       // SEC_CM: EXCEPTION.CTRL_FLOW.GLOBAL_ESC, EXCEPTION.CTRL_FLOW.LOCAL_ESC
  diff --git a/hw/top_earlgrey/data/top_earlgrey.hjson b/hw/top_earlgrey/data/top_earlgrey.hjson
  index e9422f146..6ae37d98c 100644
  --- a/hw/top_earlgrey/data/top_earlgrey.hjson
  +++ b/hw/top_earlgrey/data/top_earlgrey.hjson
  @@ -721,6 +721,7 @@
                      ICacheScramble: "1",
                      BranchPredictor: "0",
                      DbgTriggerEn: "1",
  +		   DbgHwBreakNum: "8",
                      SecureIbex: "1",
                      DmHaltAddr: "tl_main_pkg::ADDR_SPACE_RV_DM__ROM + dm::HaltAddress[31:0]",
[... 113 lines cut ...]
    (gdb) j *0x8084
    Line 91 is not in `mmio_region_write32'.  Jump anyway? (y or n) y
    Continuing at 0x8084.

    Temporary breakpoint 1, rom_test_main () at ../sw/device/lib/testing/test_rom/test_rom.c:52
    52	  CHECK_DIF_OK(dif_pinmux_init(
    (gdb) si
    0x0000823c	52	  CHECK_DIF_OK(dif_pinmux_init(
    (gdb) si
    0x0000823e	52	  CHECK_DIF_OK(dif_pinmux_init(
    (gdb) 
    0x00008242	52	  CHECK_DIF_OK(dif_pinmux_init(
    (gdb) 
    0x00008246	52	  CHECK_DIF_OK(dif_pinmux_init(
    (gdb) 
    0x0000824a	52	  CHECK_DIF_OK(dif_pinmux_init(
    (gdb) 
    Ignoring packet error, continuing...
    Invalid hex digit 116
    (gdb) 

  ```
## Question
- When I run the GDB sequences in the examples on Verilator, I consistently get the same errors. On the FPGA, everything works fine.
- I am not sure if something is wrong with my environment (e.g, openocd/GDB settings) or if there is a problem with how Verilator model handles the requests from GDB/Openocd. I just wanted to open a thread to discuss the issue and check if anyone has an idea on the root cause of it.

GregAC, 2022-04-27:
Thanks for the report. I suspect it's something to do with the Verilator/OpenOCD interface given it works fine on FPGA. I'll see if 8 hardware breakpoints works on Ibex Super System (FPGA system that gives you an Ibex core and debug and not much else, far simpler than OT so easier to debug) as a first step.

johngt, 2022-07-19:
@GregAC - not sure if there were any updates on this or if this is something that has been resolved in one-to-one communications? When you have a chance could you see if this is something that is still a live issue.

GregAC, 2022-09-22:
Quick update: I've had a look at an Ibex Super System build with 8 trigger points on FPGA and all looks to work fine. So whatever is happening here is specific to the OT Verilator infrastructure.