lowRISC/ibex #1639: Unable to build simple sytstem

jayasrikiran, 2022-04-27:
## My Environment

fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system_cosim --RV32E=0 --RV32M=ibex_pkg::RV32MFast

ERROR: In file included from ../src/lowrisc_dv_cosim_0/spike_cosim.cc:5:0:
../src/lowrisc_dv_cosim_0/spike_cosim.h:9:10: fatal error: riscv/devices.h: No such file or directory
 #include "riscv/devices.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [spike_cosim.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Vibex_simple_system] Error 2

ERROR: Failed to build lowrisc:ibex:ibex_simple_system_cosim:0 : '['make', '-j', '6']' exited with an error: 2
**EDA tool and version:**

Using Cadence 20.03-s08
**Operating system:**

CentOS 7
**Version of the Ibex source code:**

Using master branch

rswarbrick, 2022-04-27:
Looks like you're not using a supported version of Spike. See https://ibex-core.readthedocs.io/en/latest/03_reference/cosim.html for details of what you need.

rswarbrick, 2022-04-28:
Hmm, I just checked this locally and it seems to work for me, so I think it's probably a problem with your setup. I'd suggest taking a look at `build/lowrisc_ibex_ibex_simple_system_cosim_0/sim-verilator/config.mk` to figure out what's going on. Mine looks like this, for comparison:
```
#Auto generated by Edalize

TOP_MODULE        := ibex_simple_system
VC_FILE           := lowrisc_ibex_ibex_simple_system_cosim_0.vc
VERILATOR_OPTIONS := --trace --trace-fst --trace-structs --trace-params --trace-max-array 1024 -CFLAGS "-std=c++11 -Wall -DVL_USER_STOP -DVM_TRACE_FMT_FST -DTOPLEVEL_NAME=ibex_simple_system -g `pkg-config --cflags riscv-riscv riscv-disasm riscv-fdt`" -LDFLAGS "-pthread -lutil -lelf `pkg-config --libs riscv-riscv riscv-disasm riscv-fdt`" -Wall -Wwarn-IMPERFECTSCH --unroll-count 72
MAKE_OPTIONS      := 
```

johngt, 2022-07-19:
No updates from original poster, so marking the issue as pending close. If there are further queries related to this please update, otherwise this issue will be closed.