lowRISC/ibex #1645: Simple system VCS simulation fails with multiple errors

sabbari, 2022-05-02:
## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

rswarbrick, 2022-05-04:
Hmm. I agree with what you're saying, Pirmin. I'm a bit surprised if the core_uvm agents don't drive rdata to X on writes: they probably should! I think the assertion just needs fixing. @GregAC, I think you added it as part of 0f69d4972: any strong reason not to factor in "and it was a read"?

abdullahyildiz, 2022-07-16:
Hi,

I get similar errors for commit 7ba6667.

jamesrbailey, 2022-07-25:
We have run into the IbexDataRPayloadX issue as well on our own system. 

The prim_generic_ram_1p module does not initialize its `rdata_o` flops, so they are X until the first read. Therefore if you write this memory model before reading it, it will necessarily set off this assertion. As a workaround, we initialize these flops to zero.

Agreed with comments that qualifying the assertion based on it being a read is sensible. This information does not appear to be available in the ibex_top hierarchy where the assertion exists, but I *think* a hierarchical reference to `u_ibex_core.load_store_unit_i.data_we_q` would work.

GregAC, 2022-07-26:
My apologies for the slow response to this.

I have made a fix in https://github.com/lowRISC/ibex/pull/1725. I added some tracking logic so the assertions know which outstanding requests are read and then apply the X checks appropriately. This allows me to successfully run the hello_test binary on simple system under VCS. I've observed the first response, which is to a write, having Xs in rdata_i not triggering assertion failures.

We should really drive these inputs to X in the UVM testbench. I'll create a separate issue to trace.

PR #1725 (2022-07-26): [dv] Fix IbexDataRPayloadX assertion
Some aspects of the memory response are only relevant to reads. This
introduces outstanding request tracking so we know which outstanding
requests are reads and applies X checks appropriately.

Fixes #1645
commit 6dc06837 (2022-07-26): [dv] Fix IbexDataRPayloadX assertion

Some aspects of the memory response are only relevant to reads. This
introduces outstanding request tracking so we know which outstanding
requests are reads and applies X checks appropriately.

Fixes #1645
closed 2022-07-26 by GregAC