lowRISC/ibex #1656: Report a bug in Ibex

hcallahan-lowrisc, 2022-05-12:
## Observed Behavior
riscv_mem_error_test fails for IBEX_CONFIG={small,experimental-branch-predictor} when using the cosim flow

This failure is highlighted because riscv_mem_error_test was part of the smoketest ci run against all pull requests, and has been swapped out for the riscv_arithmetic_basic_test in the meantime.

## Steps to reproduce the issue
```
cd dv/uvm/core_ibex
make -k SIMULATOR=xlm ITERATIONS=1 SEED=1 TEST=riscv_mem_error_test IBEX_CONFIG=small
```

GregAC, 2022-08-11:
Sorted out passing through the PMP config in this PR: https://github.com/lowRISC/ibex/pull/1766

Though there's more to it.

- We need to make PMP CSRs illegal in non PMP configs: https://github.com/lowRISC/ibex/pull/1767
- RISC-V DV shouldn't generate PMP CSR accesses in the exception handler for non PMP configs (otherwise the mem_error test gets confused when it sees the illegal instruction handler). - https://github.com/google/riscv-dv/pull/888
- Spike needs it's PMP CSR behaviour unifying (so both PMPADDRx and PMPCFGx accesses where there are no PMP regions result in an illegal instruction trap) - https://github.com/lowRISC/riscv-isa-sim/pull/13

With that lot merged work remaining is to vendor across the new RISCV-DV and build the spike bugfix into our cosim release.

GregAC, 2022-08-31:
Further work here, riscv_mem_error_test was broken under the opentitan config. This was down to PMP. RISCV-DV sets up PMP regions by default and riscv_mem_error_test did some accesses that generated PMP access errors which then confused the test.

A fix for RISC-V DV is here: https://github.com/google/riscv-dv/pull/896

GregAC, 2022-09-06:
This has now been fixed https://github.com/lowRISC/ibex/pull/1800 (though required the removal of the branch predictor config from CI)