lowRISC/ibex #1658: Failed in generating csr test

lotharelvin, 2022-05-16:
## Observed Behavior

riscv32-unknown-elf-gcc: error: out/run/riscv_csr_test.30802/test.S: No such file or directory
riscv32-unknown-elf-gcc: fatal error: no input files

## Expected Behavior

## Steps to reproduce the issue

I entered "make" in dv/uvm/core_ibex, all the test before csr test are generated and traces are compared.
I went to the out/run/riscv_csr_test.30802/ folder and checked the riscv_csr_test.30802.riscv-dv.log, its contents are as follows:
`Mon, 16 May 2022 15:33:18 INFO     Creating output directory: /tmp/tmp74o9j11h/@@PLACEHOLDER@@
Mon, 16 May 2022 15:33:18 INFO     Processing regression test list : /home/oyykyook/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/testlist.yaml, test: riscv_csr_test
Mon, 16 May 2022 15:33:18 INFO     Found matched tests: riscv_csr_test, iterations:1
Mon, 16 May 2022 15:33:18 INFO     Processing simulator setup file : /home/oyykyook/ibex/vendor/google_riscv-dv/yaml/simulator.yaml
Mon, 16 May 2022 15:33:18 INFO     Found matching simulator: vcs
Mon, 16 May 2022 15:33:18 INFO     Running RISC-V instruction generator
Mon, 16 May 2022 15:33:18 INFO     Generating 1 riscv_csr_test`

I don't know why the .S code didn't get generated.
## My Environment

**EDA tool and version:**

**Operating system:**

**Version of the Ibex source code:**

Thanks!

rswarbrick, 2022-05-16:
Hmm, that's odd! It's definitely *supposed* to work. Could you zip up and attach the logs in the build directory?

lotharelvin, 2022-05-16:
> Hmm, that's odd! It's definitely _supposed_ to work. Could you zip up and attach the logs in the build directory?

Sure, I have attached my build folder.
 
I also found that after running a riscv_arithmetic_basic_test, the spike_cosim.log.csv is empty but it still reports _pass_. So I also attach the run folder for the arithmetic test.

I have deleted some files in the directory because they are too large, if you need any file please tell me.

Thank you!
[logs-in-build-directory.zip](https://github.com/lowRISC/ibex/files/8699010/logs-in-build-directory.zip)
[logs_in_riscv_arithmetic_basic_test.zip](https://github.com/lowRISC/ibex/files/8699012/logs_in_riscv_arithmetic_basic_test.zip)

rswarbrick, 2022-05-16:
Oh, sorry! That's not quite what I meant (I was being lazy and didn't explain myself properly).

If I run the following command:
```
make -j4 -C dv/uvm/core_ibex ITERATIONS=1 SEED=1 TEST=riscv_csr_test VERBOSE=1
```
in a clean tree (09d2dd2) then the test runs to completion. (The test is actually failing, but I'm at least getting as far as running everything).

All the files that you care about should appear in `out/run/riscv_csr_test.1` (for seed 1). My `riscv_csr_test.1.riscv-dv.log` contains:
```
Mon, 16 May 2022 12:39:47 lib.py:249   INFO     Creating output directory: /tmp/tmptd7ddcyu/@@PLACEHOLDER@@
Mon, 16 May 2022 12:39:47 lib.py:186   DEBUG    mkdir -p /tmp/tmptd7ddcyu/@@PLACEHOLDER@@/asm_test
Mon, 16 May 2022 12:39:47 lib.py:214   INFO     Processing regression test list : /src/lr/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/testlist.yaml, test: riscv_csr_test
Mon, 16 May 2022 12:39:47 lib.py:228   INFO     Found matched tests: riscv_csr_test, iterations:1
Mon, 16 May 2022 12:39:47 run.py:87    INFO     Processing simulator setup file : /src/lr/ibex/vendor/google_riscv-dv/yaml/simulator.yaml
Mon, 16 May 2022 12:39:47 run.py:92    INFO     Found matching simulator: vcs
Mon, 16 May 2022 12:39:47 run.py:269   INFO     Running RISC-V instruction generator
Mon, 16 May 2022 12:39:47 run.py:273   INFO     Generating 1 riscv_csr_test
Mon, 16 May 2022 12:39:47 lib.py:100   DEBUG    python3 /src/lr/ibex/vendor/google_riscv-dv/scripts/gen_csr_test.py --csr_file /src/lr/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/csr_description.yaml --xlen 32 --iterations 1 --out /tmp/tmptd7ddcyu/@@PLACEHOLDER@@/asm_test --end_signature_addr 0
```

The `@@PLACEHOLDER@@` stuff is because of an ugly hack that we do to force riscv-dv to generate files following the directory layout that we want: see `scripts/run-instr-gen.py` for details. The `riscv_csr_test` test is special because riscv-dv generates slightly different files for it.

To figure out what's going on, could you edit that script and add the following lines to the start of the `do_file_copies()` function?
```python3
    print()
    print('\n'.join(os.listdir(src_dir)))
    print()
    assert 0
```
Re-running `make`, I see the following:
```
scripts/run-instr-gen.py          \
  --verbose                            \
  --simulator vcs               \
  --end-signature-addr 8ffffffc \
  --output-dir out/run/riscv_csr_test.1                     \
  --gen-build-dir out/build/instr-gen \
  --ibex-config opentitan           \
  --test-dot-seed riscv_csr_test.1
+ python3 /src/lr/ibex/vendor/google_riscv-dv/run.py --testlist /src/lr/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/testlist.yaml --gcc_opts=-mno-strict-align --custom_target /src/lr/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension --csr_yaml /src/lr/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/csr_description.yaml --mabi=ilp32 --verbose --so --steps=gen --output /tmp/tmpu8_e91q7/@@PLACEHOLDER@@ --simulator vcs --isa rv32imcb --test riscv_csr_test --start_seed 1 --iterations 1 --sim_opts '+uvm_set_inst_override=riscv_asm_program_gen,ibex_asm_program_gen,uvm_test_top.asm_gen +signature_addr=8ffffffc +pmp_num_regions=16 +pmp_granularity=0 +tvec_alignment=8' --debug /tmp/tmpu8_e91q7/cmds.list >out/run/riscv_csr_test.1/riscv_csr_test.1.riscv-dv.log 2>&1
+ python3 /src/lr/ibex/vendor/google_riscv-dv/scripts/gen_csr_test.py --csr_file /src/lr/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/csr_description.yaml --xlen 32 --iterations 1 --out /tmp/tmpu8_e91q7 --end_signature_addr 0 >/dev/null 2>&1

cmds.list
@@PLACEHOLDER@@
riscv_csr_test_0.S

ERROR: 
make: *** [Makefile:238: out/run/riscv_csr_test.1/test.S] Error 1
```

I'm assuming that you won't have `riscv_csr_test_0.S` there?

lotharelvin, 2022-05-16:
> Oh, sorry! That's not quite what I meant (I was being lazy and didn't explain myself properly).
> 
> If I run the following command:
> 
> ```
> make -j4 -C dv/uvm/core_ibex ITERATIONS=1 SEED=1 TEST=riscv_csr_test VERBOSE=1
> ```
> 
> in a clean tree ([09d2dd2](https://github.com/lowRISC/ibex/commit/09d2dd23061a12b2eaf68815e1dc1d5488fb805a)) then the test runs to completion. (The test is actually failing, but I'm at least getting as far as running everything).
> 
> All the files that you care about should appear in `out/run/riscv_csr_test.1` (for seed 1). My `riscv_csr_test.1.riscv-dv.log` contains:
> 
> ```
> Mon, 16 May 2022 12:39:47 lib.py:249   INFO     Creating output directory: /tmp/tmptd7ddcyu/@@PLACEHOLDER@@
> Mon, 16 May 2022 12:39:47 lib.py:186   DEBUG    mkdir -p /tmp/tmptd7ddcyu/@@PLACEHOLDER@@/asm_test
> Mon, 16 May 2022 12:39:47 lib.py:214   INFO     Processing regression test list : /src/lr/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/testlist.yaml, test: riscv_csr_test
> Mon, 16 May 2022 12:39:47 lib.py:228   INFO     Found matched tests: riscv_csr_test, iterations:1
> Mon, 16 May 2022 12:39:47 run.py:87    INFO     Processing simulator setup file : /src/lr/ibex/vendor/google_riscv-dv/yaml/simulator.yaml
> Mon, 16 May 2022 12:39:47 run.py:92    INFO     Found matching simulator: vcs
> Mon, 16 May 2022 12:39:47 run.py:269   INFO     Running RISC-V instruction generator
> Mon, 16 May 2022 12:39:47 run.py:273   INFO     Generating 1 riscv_csr_test
> Mon, 16 May 2022 12:39:47 lib.py:100   DEBUG    python3 /src/lr/ibex/vendor/google_riscv-dv/scripts/gen_csr_test.py --csr_file /src/lr/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/csr_description.yaml --xlen 32 --iterations 1 --out /tmp/tmptd7ddcyu/@@PLACEHOLDER@@/asm_test --end_signature_addr 0
> ```
> 
> The `@@PLACEHOLDER@@` stuff is because of an ugly hack that we do to force riscv-dv to generate files following the directory layout that we want: see `scripts/run-instr-gen.py` for details. The `riscv_csr_test` test is special because riscv-dv generates slightly different files for it.
> 
> To figure out what's going on, could you edit that script and add the following lines to the start of the `do_file_copies()` function?
> 
> ```python
>     print()
>     print('\n'.join(os.listdir(src_dir)))
>     print()
>     assert 0
> ```
> 
> Re-running `make`, I see the following:
> 
> ```
> rupert@halibut /s/l/i/d/u/core_ibex (work) > make -j4 ITERATIONS=1 SEED=1 TEST=riscv_csr_test VERBOSE=1
> scripts/run-instr-gen.py          \
>   --verbose                            \
>   --simulator vcs               \
>   --end-signature-addr 8ffffffc \
>   --output-dir out/run/riscv_csr_test.1                     \
>   --gen-build-dir out/build/instr-gen \
>   --ibex-config opentitan           \
>   --test-dot-seed riscv_csr_test.1
> + python3 /src/lr/ibex/vendor/google_riscv-dv/run.py --testlist /src/lr/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/testlist.yaml --gcc_opts=-mno-strict-align --custom_target /src/lr/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension --csr_yaml /src/lr/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/csr_description.yaml --mabi=ilp32 --verbose --so --steps=gen --output /tmp/tmpu8_e91q7/@@PLACEHOLDER@@ --simulator vcs --isa rv32imcb --test riscv_csr_test --start_seed 1 --iterations 1 --sim_opts '+uvm_set_inst_override=riscv_asm_program_gen,ibex_asm_program_gen,uvm_test_top.asm_gen +signature_addr=8ffffffc +pmp_num_regions=16 +pmp_granularity=0 +tvec_alignment=8' --debug /tmp/tmpu8_e91q7/cmds.list >out/run/riscv_csr_test.1/riscv_csr_test.1.riscv-dv.log 2>&1
> + python3 /src/lr/ibex/vendor/google_riscv-dv/scripts/gen_csr_test.py --csr_file /src/lr/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/csr_description.yaml --xlen 32 --iterations 1 --out /tmp/tmpu8_e91q7 --end_signature_addr 0 >/dev/null 2>&1
> 
> cmds.list
> @@PLACEHOLDER@@
> riscv_csr_test_0.S
> 
> ERROR: 
> make: *** [Makefile:238: out/run/riscv_csr_test.1/test.S] Error 1
> ```
> 
> I'm assuming that you won't have `riscv_csr_test_0.S` there?

Yes, after adding that lines to the `do_file_copies()` function and re-running make, I don't see `riscv_csr_test_0.S`.
My output is as follow:
`(base) [oyykyook@mics-eda ibex]$ make -j4 -C dv/uvm/core_ibex ITERATIONS=1 SEED=1 TEST=riscv_csr_test VERBOSE=1
make: Entering directory '/home/oyykyook/ibex/dv/uvm/core_ibex'
scripts/compile-tb.py             \
  --verbose                            \
  --ibex-config opentitan           \
  --output=out/build/tb               \
  --shared-cov-dir=out/run/shared_cov \
  --simulator=vcs               \
   --en_wave --en_cosim
scripts/run-instr-gen.py          \
  --verbose                            \
  --simulator vcs               \
  --end-signature-addr 8ffffffc \
  --output-dir out/run/riscv_csr_test.1                     \
  --gen-build-dir out/build/instr-gen \
  --ibex-config opentitan           \
  --test-dot-seed riscv_csr_test.1
+ python3 /home/oyykyook/ibex/vendor/google_riscv-dv/run.py --testlist /home/oyykyook/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/testlist.yaml --gcc_opts=-mno-strict-align --custom_target /home/oyykyook/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension --csr_yaml /home/oyykyook/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/csr_description.yaml --mabi=ilp32 --verbose --so --steps=gen --output /tmp/tmpnzifgitv/@@PLACEHOLDER@@ --simulator vcs --isa rv32imcb --test riscv_csr_test --start_seed 1 --iterations  [... cut ...]
+ python3 /home/oyykyook/ibex/vendor/google_riscv-dv/scripts/gen_csr_test.py --csr_file /home/oyykyook/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/csr_description.yaml --xlen 32 --iterations 1 --out /tmp/tmpnzifgitv --end_signature_addr 0 >/dev/null 2>&1

cmds.list
@@PLACEHOLDER@@

ERROR: 
make: *** [Makefile:238: out/run/riscv_csr_test.1/test.S] Error 1
make: *** Waiting for unfinished jobs....
make: *** [Makefile:331: out/build/tb/.compile.stamp] Error 2
make: Leaving directory '/home/oyykyook/ibex/dv/uvm/core_ibex'
`
Thank you!

rswarbrick, 2022-05-16:
How annoying! Well, this is kind of my fault :-( If you look at line 97 of `run-instr-gen.py`, you'll find this:
```python3
            ret = run_one(args.verbose, cmd, redirect_stdstreams='/dev/null')
```
Can you change it to this?
```python3
            ret = run_one(args.verbose, cmd)
```
Then you should at least see the error messages on your console.

I'll open a PR in a minute that writes them somewhere a bit more helpful.

rswarbrick, 2022-05-16:
For a less hacky approach, see: https://github.com/lowRISC/ibex/pull/1660

lotharelvin, 2022-05-16:
Yes I changed the line 97 in the `run-inst-gen.py` and re-ran `make -j4 -C dv/uvm/core_ibex ITERATIONS=1 SEED=1 TEST=riscv_csr_test VERBOSE=1`, first it raised an error:
`ERROR:root:Please install bitstring package: sudo apt-get install python3-bitstring
`
Then I used pip to install bitstring and ran the make again, this time the terminal output is as follows:
(base) [oyykyook@mics-eda ibex]$ make -j4 -C dv/uvm/core_ibex ITERATIONS=1 SEED=1 TEST=riscv_csr_test VERBOSE=1
make: Entering directory '/home/oyykyook/ibex/dv/uvm/core_ibex'
scripts/compile-tb.py             \
  --verbose                            \
  --ibex-config opentitan           \
  --output=out/build/tb               \
  --shared-cov-dir=out/run/shared_cov \
  --simulator=vcs               \
   --en_wave --en_cosim
scripts/run-instr-gen.py          \
  --verbose                            \
  --simulator vcs               \
  --end-signature-addr 8ffffffc \
  --output-dir out/run/riscv_csr_test.1                     \
  --gen-build-dir out/build/instr-gen \
  --ibex-config opentitan           \
  --test-dot-seed riscv_csr_test.1
+ python3 /home/oyykyook/ibex/vendor/google_riscv-dv/run.py --testlist /home/oyykyook/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/testlist.yaml --gcc_opts=-mno-strict-align --custom_target /home/oyykyook/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension --csr_yaml /home/oyykyook/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/csr_description.yaml --mabi=ilp32 --verbose --so --steps=gen --output /tmp/tmpuob8sbng/@@PLACEHOLDER@@ --simulator vcs --isa rv32imcb --test riscv_csr_test --start_seed 1 --iterations 1 --sim_opts '+uvm_set_inst_override=riscv_asm_program_gen,ibex_asm_program_gen,uvm_test_top.asm_gen +signature_addr=8ffffffc +pmp_num_regions=16 +pmp_granularity=0 +tvec_alignment=8' --debug /tmp/tmpuob8sbng/cmds.list >out/run/riscv_csr_test.1/riscv_csr_test.1.riscv-dv.log 2>&1
+ sh -c 'vcs -f /home/oyykyook/ibex/dv/uvm/core_ibex/ibex_dv.f  -full64 -l out/build/tb/compile.log -sverilog -ntb_opts uvm-1.2 +define+UVM +define+UVM_REGEX_NO_DPI -timescale=1ns/10ps -licqueue -LDFLAGS '"'"'-Wl,--no-as-needed'"'"' -CFLAGS '"'"'--std=c99 -fno-extended-identifiers'"'"' -Mdir=out/build/tb/vcs_simv.csrc -o out/build/tb/vcs_simv -debug_access+pp -xlrm uniq_prior_final -CFLAGS '"'"'--std=c99 -fno-extended-identifiers'"'"' -lca -kdb -pvalue+core_ibex_tb_top.RV32E=0 +define+IBEX_CFG_RV32M=ibex_pkg::RV32MSingleCycle +define+IBEX_CFG_RV32B=ibex_pkg::RV32BOTEarlGrey +define+IBEX_CFG_RegFile=ibex_pkg::RegFileFF -pvalue+core_ibex_tb_top.BranchTargetALU=1 -pvalue+core_ibex_tb_top.WritebackStage=1 -pvalue+core_ibex_tb_top.ICache=1 -pvalue+core_ibex_tb_top.ICacheECC=1 -pvalue+core_ibex_tb_top.ICacheScramble=1 -pvalue+core_ibex_tb_top.BranchPredictor=0 -pvalue+core_ibex_tb_top.DbgTriggerEn=1 -pvalue+core_ibex_tb_top.SecureIbex=1 -pvalue+core_ibex_tb_top.PMPEnable=1 -pvalue+core_ibex_tb_top.PMPGranularity=0 -pvalue+core_ibex_tb_top.PMPNumRegions=16 -pvalue+core_ibex_tb_top.MHPMCounterNum=10 -pvalue+core_ibex_tb_top.MHPMCounterWidth=32 -debug_access+all -ucli -do vcs.tcl   -f /home/oyykyook/ibex/dv/uvm/core_ibex/ibex_dv_cosim_dpi.f +define+INC_IBEX_COSIM -LDFLAGS '"'"'-Wl,-rpath,/home/oyykyook/ibex_toolchain/lowrisc-toolchain-gcc-rv32imcb-20220210-1/lib'"'"' -CFLAGS '"'"'-I/home/oyykyook/ibex_toolchain/lowrisc-toolchain-gcc-rv32imcb-20220210-1/include -I/home/oyykyook/ibex_toolchain/lowrisc-toolchain-gcc-rv32imcb-20220210-1/include/softfloat'"'"' -CFLAGS '"'"'-I/home/oyykyook/ibex/dv/cosim'"'"' -lriscv -lsoftfloat -ldl -ldisasm -lfdt -lstdc++ ' >/dev/null 2>&1
+ python3 /home/oyykyook/ibex/vendor/google_riscv-dv/scripts/gen_csr_test.py --csr_file /home/oyykyook/ibex/dv/uvm/core_ibex/scripts/../riscv_dv_extension/csr_description.yaml --xlen 32 --iterations 1 --out /tmp/tmpuob8sbng --end_signature_addr 0

cmds.list
@@PLACEHOLDER@@
riscv_csr_test_0.S

ERROR: 
make: *** [Makefile:238: out/run/riscv_csr_test.1/test.S] Error 1
make: *** Waiting for unfinished jobs....
make: *** [Makefile:331: out/build/tb/.compile.stamp] Error 2
make: Leaving directory '/home/oyykyook/ibex/dv/uvm/core_ibex'

I can see there is the .S file now, but the compile failed though, so I checked the /build/tb/compile.log and I found following errors:
`/home/oyykyook/ibex/dv/cosim/spike_cosim.cc:301:14: error: 'class processor_t' has \
no member named 'set_csr'; did you mean 'get_csr'?
  301 |   processor->set_csr(CSR_MSTATUS, mstatus);
      |              ^~~~~~~
      |              get_csr
/home/oyykyook/ibex/dv/cosim/spike_cosim.cc:303:14: error: 'class processor_t' has \
no member named 'set_csr'; did you mean 'get_csr'?
  303 |   processor->set_csr(CSR_MEPC, mstack.epc);
      |              ^~~~~~~
      |              get_csr
/home/oyykyook/ibex/dv/cosim/spike_cosim.cc:304:14: error: 'class processor_t' has \
no member named 'set_csr'; did you mean 'get_csr'?
  304 |   processor->set_csr(CSR_MCAUSE, mstack.cause);
      |              ^~~~~~~
      |              get_csr
/home/oyykyook/ibex/dv/cosim/spike_cosim.cc: In member function 'void SpikeCosim::fixup_csr(int, \
uint32_t)':
/home/oyykyook/ibex/dv/cosim/spike_cosim.cc:365:18: error: 'class processor_t' has \
no member named 'set_csr'; did you mean 'get_csr'?
  365 |       processor->set_csr(csr_num, new_val);
      |                  ^~~~~~~
      |                  get_csr
In file included from /home/oyykyook/centos/riscv-gcc/include/c++/10.2.0/memory:83, \

[... cut ...]
  234 |   processor_t(const isa_parser_t *isa, const char* varch,
      |   ^~~~~~~~~~~
/home/oyykyook/ibex_toolchain/lowrisc-toolchain-gcc-rv32imcb-20220210-1/include/riscv/processor.h:234:3: \
note:   candidate expects 7 arguments, 8 provided
make[2]: *** [filelist:19: spike_cosim.o] Error 1
make[1]: *** [Makefile:111: product_clean_order] Error 2
make[1]: Leaving directory '/home/oyykyook/ibex/dv/uvm/core_ibex/out/build/tb/vcs_simv.csrc' \
`
So do I need to change `set_csr` to `get_csr`?
Thank you for your help!

lotharelvin, 2022-05-16:
> 

Hi I have changed my code according to this PR, but when I run make, I still got some problems:
`Traceback (most recent call last):
  File "/home/oyykyook/ibex/dv/uvm/core_ibex/scripts/compile-tb.py", line 104, in <module>
    sys.exit(main())
  File "/home/oyykyook/ibex/dv/uvm/core_ibex/scripts/compile-tb.py", line 95, in main
    retcode = run_one(args.verbose, ['sh', '-c', cmd],
  File "/home/oyykyook/ibex/dv/uvm/core_ibex/scripts/scripts_lib.py", line 68, in run_one
    print('+ ' + cmd_str, file=stdstream_dest)
AttributeError: 'int' object has no attribute 'write'
make: *** [Makefile:331: out/build/tb/.compile.stamp] Error 1
`
Thanks for your work!

rswarbrick, 2022-05-16:
The C++ build problems look a lot like you're using an unsupported version of Spike. I think we now default to enabling "co-simulation". The easiest approach is to install the `lowrisc-cosim-v0.2` version, as described [here](https://ibex-core.readthedocs.io/en/latest/03_reference/cosim.html). I think we forgot to update some documents: oops!

You might also have some success with specifying COSIM=0 on your Make command line, but we're planning to remove that option (and sort out the documentation) in the next few days.

lotharelvin, 2022-05-17:
Hi, sorry for the late reply.
I have fixed spike problem and changed my code according to #1660 .

I ran `make -j4  ITERATIONS=1 SEED=1 TEST=riscv_csr_test VERBOSE=1`, then in `/out/run/riscv_csr_test.1/test-result.yml` I found:
`name:            riscv_csr_test
seed:            1
binary:          out/run/riscv_csr_test.1/test.o
uvm_log:         out/run/riscv_csr_test.1/rtl.log
rtl_trace:       out/run/riscv_csr_test.1/trace_core_00000000.log
rtl_trace_csv:   out/run/riscv_csr_test.1/trace_core_00000000.log.csv
iss_trace:
iss_trace_csv:
en_cosim:        True
cosim_trace:
cosim_trace_csv:
comparison_log:
passed:          False
failure_message: |-
  UVM_ERROR /home/oyykyook/ibex/vendor/lowrisc_ip/dv/sv/mem_model/mem_model.sv(35) @ 4378400: reporter [mem] read from uninitialized addr 0x80000b5f
  [FAILED]: sim error seen
`
Then I checked rtl.log, and it reports following info:
`
UVM_INFO @ 0: reporter [RNTST] Running test core_ibex_csr_test...
UVM_INFO /home/oyykyook/ibex/dv/uvm/core_ibex/env/core_ibex_env.sv(37) @ 0: uvm_test_top.env [uvm_test_top.env] Cosim disabled by plusarg +disable_cosim=1
UVM_INFO /home/oyykyook/ibex/dv/uvm/core_ibex/tests/core_ibex_base_test.sv(169) @ 202400: uvm_test_top [uvm_test_top] Running test : out/run/riscv_csr_test.1/test.bin
UVM_INFO /home/oyykyook/ibex/dv/uvm/core_ibex/common/ibex_mem_intf_agent/ibex_mem_intf_response_seq_lib.sv(26) @ 202400: uvm_test_top.env.instr_if_response_agent.sequencer@@instr_intf_seq [uvm_test_top.env.instr_if_response_agent.sequencer.instr_intf_seq] is_dmem_seq: 0x0
UVM_INFO /home/oyykyook/ibex/dv/uvm/core_ibex/common/ibex_mem_intf_agent/ibex_mem_intf_response_seq_lib.sv(26) @ 202400: uvm_test_top.env.data_if_response_agent.sequencer@@data_intf_seq [uvm_test_top.env.data_if_response_agent.sequencer.data_intf_seq] is_dmem_seq: 0x1
core_ibex_tb_top.dut.u_ibex_tracer.printbuffer_dumpline.unnamed$$_0: Writing execution trace to out/run/riscv_csr_test.1/trace_core_00000000.log
UVM_ERROR /home/oyykyook/ibex/vendor/lowrisc_ip/dv/sv/mem_model/mem_model.sv(35) @ 4378400: reporter [mem] read from uninitialized addr 0x80000b5f

--- RISC-V UVM TEST FAILED ---

`
Looks like we have found the cause of the problem, but I don't have much experience using vcs so I still don't know how to fix it.
Thanks for your help!

rswarbrick, 2022-05-17:
Ahah! That's excellent: you're now at the same point that we are :-) Sadly, this specific test is currently failing. Fortunately, I think I can see what's going on: the memory model is overly zealous about making sure we don't read from uninitialised addresses. It turns out that the test is executing an instruction near the top of initialised IMEM and we're reading a couple of words ahead. Oops!

I'm going to take a proper look either this evening or tomorrow.

rswarbrick, 2022-05-17:
As a follow-up, I've done a bit of debugging and fixed one DV bug (https://github.com/lowRISC/ibex/pull/1662). Unfortunately, this wasn't the root cause of why the test is failing. Hopefully I'll get to that soon.

rswarbrick, 2022-05-17:
A tiny bit more debugging (mostly to help me pick things up tomorrow morning): The problem is that we describe the `mcause` CSR with the following hunk of YAML:
```
# MCAUSE
- csr: mcause
  description: >
    Indicates trap cause
  address: 0x342
  privilege_mode: M
  rv32:
    - field_name: Interrupt
      description: >
        Indicates if trap caused by interrupt
      type: WARL
      reset_val: 0
      msb: 31
      lsb: 31
    - field_name: Exception Code
      type: WLRL
      reset_val: 0
      msb: 4
      lsb: 0
```

This might have been right once, but is no longer correct since Greg implemented internal interrupts. I don't think that `gen_csr_test.py` supports the behaviour that we've got for our MCAUSE field, which boils down to this code in `ibex_cs_registers.sv`:
```
      // mcause: exception cause
      CSR_MCAUSE: csr_rdata_int = {mcause_q.irq_ext | mcause_q.irq_int,
                                   mcause_q.irq_int ? {26{1'b1}} : 26'b0,
                                   mcause_q.lower_cause[4:0]};
```

Here, we ostensibly have a 31-bit exception code field, but the top bits are all constrained to be equal.

@GregAC: What do you think we should do? A lame response might be to just use 6 bits and represent internal interrupts starting at 32, without the bit extension. Or we could split the field and just set the top bit? Or we could do some work to generalise the DV code to understand this sort of thing: for example, we could say that the field is 6 bits but the other 25 bits are some sort of mirror. But that means some DV work and (eek) getting a patch into riscv-dv.

rswarbrick, 2022-05-18:
Right, I think we've got to the stage where everything is building for you. And I've fixed the silly bug in the script that meant we weren't exposing error messages (oops). I'm going to close this now and track the failing test in its own issue (linked just above). Thank you very much for the bug report.

closed 2022-05-18 by rswarbrick